waveshareteam/e-Paper #126: Unable to compile c examples

parkerlreed, 2021-01-02:
Arch Linux ARM

```
[parker@dslrpi c]$ make
gcc -g -O0 -Wall -D USE_WIRINGPI_LIB -D RPI -c  ./lib/Config/dev_hardware_SPI.c -o ./bin/dev_hardware_SPI.o -lwiringPi -lm  -D DEBUG
Assembler messages:
Fatal error: can't create ./bin/dev_hardware_SPI.o: No such file or directory
make: *** [Makefile:69: RPI_DEV] Error 1
```

parkerlreed, 2021-01-02:
c Makefile expects ./bin/ to exist but doesn't create it if missing. And it's not in the repo itself.

SSYYL, 2021-01-04:
Try using the root to make.
eg: sudo make

parkerlreed, 2021-01-04:
root doesn't make any difference here. The ./bin directory is defined but never checked to see if it exists/created. https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/c/Makefile#L6

SSYYL, 2021-01-05:
We have not considered the problem, Because normally this directory exists.
 you can create the ./bin manually.

closed 2021-02-20 by SSYYL