waveshareteam/e-Paper #134: Please add support for the 880×528, 7.5inch E-Ink display HAT for Raspberry Pi, red / black / white three-color

homeallone, 2021-02-04:
Currently the drivers don't support this HD color version. Could you please add a driver for this?

toothrot, 2021-02-19:
I am having this same issue. I see it listed on the wiki page for Python and C, but not in the repo:

https://www.waveshare.com/wiki/7.5inch_HD_e-Paper_HAT_(B)
```
epd_7in5bc_V2_test.py：Example for 7.5inch e-paper B V2（Black/White/Red）; This is the current version with V2 sticker on the backside. (2020-07-29);
```

SSYYL, 2021-02-19:
C examples: [click](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/c/examples/EPD_7in5b_HD_test.c)
Python example: [click](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/examples/epd_7in5b_HD_test.py)

They've always been here

toothrot, 2021-02-19:
Those are the ones I tried, but I have this issue:

```
$ sudo ./epd 
EPD_7IN5B_HD_test Demo
/***********************************/ 
Current environment: Raspbian
set wiringPi lib success !!! 
/***********************************/ 
e-Paper Init and Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
4 S
NewImage:BlackImage and RYImage
show window BMP-----------------
pixel = 100 * 100
pixel = 100 * 100
Debug: e-Paper busy
Debug: e-Paper busy release
show red bmp------------------------
pixel = 880 * 528
pixel = 880 * 528
Debug: e-Paper busy
Debug: e-Paper busy release
EPD_Display
Debug: e-Paper busy
Debug: e-Paper busy release
Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Goto Sleep...
close 5V, Module enters 0 power consumption ...
```

The display stays white. It is brand new and undamaged. I have tried reconnecting all ribbon wires, and the HAT. Raspberry Pi 3b is also brand new, and has a fresh install of Raspbian today. I followed wiki instructions for all the required libraries, but I can't figure this out.

I'm happy to help debug.

SSYYL, 2021-02-19:
I've just tried the program, and it works.
EPD is right. Can you take a picture of the device connection and the version of the e-Paper Driver HAT?

homeallone, 2021-02-19:
Hi alexander,

I finally got it working with the epd7in5_V2.py
<https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd7in5_V2.py>,
which you don't want (black and white) ... but it worked... both c and
python
I had remove the following lines in the config file of python to get the
python part running

epdconfig.py
64,66d63
<     def spi_writebyte2(self, data):
<         self.SPI.writebytes2(data)
<

So I still don't understand.. and if I compare the datasheet with the way
this driver handles the display. It cannot be the one which waveshare is
thinking it should be.

toothrot, 2021-02-19:
@SSYYL Here are the pictures: https://imgur.com/a/xPFfZVt

@homeallone Thanks, but that doesn't seem to work for me.

homeallone, 2021-02-19:
Did you rebuild your libs
~/e-Paper/RaspberryPi_JetsonNano/python
sudo python3 setup.py build
sudo python3 setup.py install

And in c
I have added the Makefile
USELIB_RPI = USE_BCM2835_LIB
and the main.c in examples
        EPD_7in5_V2_test(); // Werkt!

SSYYL, 2021-02-19:
@toothrot , Have you ever tried to connect PI and HAT using the PH2.0 8PIN connection block.
If not, please try it.

toothrot, 2021-02-19:
@SSYYL Thanks! The following pinout worked great with no code modifications:

```
VCC  - 3.3v      - Pin 1
GND  - Ground    - Pin 9
DIN  - SPI0 MOSI - Pin 19 (GPIO 10)
CLK  - SPI0 SCLK - Pin 23 (GPIO 11)
CS   - SPI0 CE0  - Pin 24 (GPIO 8)
DC   - Data/Cmd  - Pin 22 (GPIO 25)
RST  - Reset     - Pin 11 (GPIO 17)
Busy - Busy      - Pin 18 (GPIO 24)
```

@homeallone I think my issue was with the GPIO on the board, but manually connecting with the provided pins worked great for the driver code they recommended.

SSYYL, 2021-02-20:
@toothrot , I'm glad it worked.

homeallone, 2021-02-21:
@toothrot can you tell me your SKU number?

peterquax, 2021-02-21:
I'm having the same exact issue. Tried with 2 displays (same SKU, 17960 and HAT 2.1) and two raspberry PI's (both 3B and Pi Zero W). I tried connecting both the HAT directly to the RPI as well as through the header cables, makes no difference at all. Like homeallone, I can get the epd7in5_V2.py example working fine, but that one has no color and is not HD.  I tried both the python and C examples of EPD_7IN5B_HD_test. They both hang at the same moment : 

EPD_7IN5B_HD_test Demo
/***********************************/
Current environment: Raspbian
set wiringPi lib success !!!
/***********************************/
e-Paper Init and Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy

The display always stays blank. 

Any other suggestions to try ?

SSYYL, 2021-02-22:
Hi ,@peterquax @homeallone 
Theoretically it is absolutely impossible for 7.5_V2 and 7.5b_HD to drive the same EPD.
Please confirm whether the EPD in your hand is the same as the picture of @toothrot.

peterquax, 2021-02-22:
I've checked both my displays and the writing on the flex cable is indeed different, as well as the color of the screen protector tab. I've attached some pics below. When looking at the display under the microscope, I seem to be able to read the following ID : 075RSE01 (backwards). However the packaging clearly mentions the right SKU for the HD model. Is there a way to verify for sure that the display matches the box ?

[image]
[image]
[image]

SSYYL, 2021-02-22:
The FPC shows 5.83 V2. Are you sure this screen is in the box? The size is not 7.5inch. Please contact the customer service.

homeallone, 2021-02-22:
Hi Peter,

I have indeed exactly the same model! In the exact same box. I bought it
via a Dutch a dutch vendor. I already emailed the vendor that I
suspect it's not the right model. But they did not reply.

@SSYYL  Any advice for me?

peterquax, 2021-02-22:
Hi Homeallone. 

Would that be kiwi-electronics.nl by any chance ? If so, we've got the exact same issue :( Sending them an e-mail right now.

homeallone, 2021-02-22:
Yes, it is.

peterquax, 2021-02-22:
Hi @SSYYL,

Yes, boxes seemed to be unopened and all contents were intact (bought two displays at once). They are absolutely 7.5 inch, just measured them. Did you mean to contact waveshare support ? If I return them to the vendor, they might just send me two new boxes with the same issue (if they got a wrong batch).

homeallone, 2021-02-22:
Peter,

Please refer me in your email, My incident number is: #26119
<https://service.kiwi-electronics.nl/hc/requests/26119> Two stands stronger
then one.. :)

SSYYL, 2021-02-22:
Contact the dealer or waveshare customer service, Make sure the EPD is correct.

peterquax, 2021-02-22:
It seems like the distributor knows about the issue and is offering an exchange, so hopefully that will resolve our issues :) Thanks to all for the help offered !

closed 2021-05-06 by SSYYL