waveshareteam/e-Paper #142: EPD_HEIGHT and EPD_WIDTH values are inverted in epd2in13* drivers

fileinster, 2021-03-01:
I only checked the 2in13 drivers, but all of them have the height and width values inverted, i.e., the height is the width and vice-versa.

Not an issue if you know about it, but worth correcting in a new release.

SSYYL, 2021-03-02:
There is no inverse if you follow the default Settings(The default is FPC on top), and in the program is easy to rotate.
So we don't have a strict distinction between height and width (and probably don't need to).

KKoovalsky, 2021-05-05:
@SSYYL then when having a rotated image how to configure the display to increment with Y axis and not the X axis?

I use EPD 2.13 Black & White, in version 2.13.

I can't get it working, this is what I do:

1. Set the `Data Entry Mode` to `0b101`.
2. Call `EPD_2IN13_V2_Display` with image aligned by Y pixels.
3. Get rubbish on the screen.

I couldn't find any documentation about the possible image update sequences and how to apply them. Could you provide one?

KKoovalsky, 2021-05-12:
@SSYYL Could you take a look? :)

SSYYL, 2021-05-12:
@KKoovalsky , Hello, I don't quite understand what you mean, could you show me the schematic diagram?

KKoovalsky, 2021-05-12:
You mean the hardware schematic? I use this one: https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT

Let me explain my issue in a more detailed way. 

In the product I currently develop, the image is provided as row-by-row data, one bit per pixel. The thing is that, as mentioned in this issue, width and height are inverted for this display. So the width is 122 px and height is 250 px. The image format, which is provided to me, has width 250 px and 122 px height.

[image]
 
The problem here is that each byte contains pixel aligned in rows, not by columns. Waveshare EPD library (this library) expects the pixels to be aligned by columns (green boxes, representing a byte with 8 pixels). I receive an image which is aligned by rows (blue boxes). This means that I have to rotate the image, which is really nasty, because I need to iterate over single pixels in a vector (apply offsets, shifting, masking, etc. - complex stuff). 

My question was: _Can I configure the EPD library so that it renders images that are row-aligned?_

I tried to configure the "Data Entry mode setting":

[image]

And changed the default `AM` bit to `1`, but it didn't work as expected.

Could you help me with the question? :)

SSYYL, 2021-05-12:
Okay, got it. Several registers need to be modified and i need to run some tests.

SSYYL, 2021-05-13:
EPD_2IN13_V2_SendCommand(0x11); //data entry mode
        EPD_2IN13_V2_SendData(0x04);

        EPD_2IN13_V2_SendCommand(0x44); //set Ram-X address start/end position
        EPD_2IN13_V2_SendData(0x0f);
        EPD_2IN13_V2_SendData(0x00);   
        EPD_2IN13_V2_SendCommand(0x45); //set Ram-Y address start/end position
        EPD_2IN13_V2_SendData(0xF9);   //0xF9-->(249+1)=250
        EPD_2IN13_V2_SendData(0x00);
        EPD_2IN13_V2_SendData(0x00);
        EPD_2IN13_V2_SendData(0x00);

        EPD_2IN13_V2_SendCommand(0x4E);   // set RAM x address count to 0;
        EPD_2IN13_V2_SendData(0x0f);
        EPD_2IN13_V2_SendCommand(0x4F);   // set RAM y address count to 0X127;
        EPD_2IN13_V2_SendData(0xF9);
        EPD_2IN13_V2_SendData(0x00);

I'm not quite sure if this works, because I found that the image is rotated and mirrored.
[image]
You need to adjust these three sets of registers, 0x11 and 0x44 & 0x45 and 0x4E & 0x4F.