waveshareteam/e-Paper #145: 1.54 and 2.9 Not compiling in Arduino IDE 1.8.13

nnc6, 2021-03-08:
As in the title this examples are not compiling in Arduino IDE. I got:
avr/pgmspace.h: No such file or directory
I found [https://www.arduino.cc/reference/en/language/variables/utilities/progmem/](url) which says this library is already included in modern IDE so I tried to comment #include <avr/pgmspace.h> and I got: 
pgm_read_byte' was not declared in this scope
I would appreciate some informations about how to make this displays work!

SSYYL, 2021-03-11:
This header file should already be included in the project and is shipped with the IDE by default.
Please add the detailed error information.

JaimieD, 2021-04-13:
I have come across this issue and resolved it temporarily by adding the library to the working folder and getting rid of the "avr/" in my "include" lines.