waveshareteam/e-Paper #147: Python support for 13.3 inch e-paper display

TalinR, 2021-03-11:
Hi just wondering if python support will be added for the 13.3 inch epaper display? Otherwise is it possible to modify resolution values of the other displays and use the 13 inch one? Thanks

SSYYL, 2021-03-11:
Hello, 13.3inch e-Paper(SPI) is a beta version, and will be available in Python if it is available in our store in the future.

TalinR, 2021-03-11:
Sorry do you know when approximately this may be? The product is already available in your store and says nothing about being a beta version https://www.waveshare.com/13.3inch-e-paper-hat.htm

SSYYL, 2021-03-12:
Sorry, I thought you said SPI interface.
It may take a while, but there are customers who have developed Python versions that you can look up on GitHub

closed 2021-05-06 by SSYYL