waveshareteam/e-Paper #154: Screen Tearing - Epd_7in5_V2 with Raspberry Pi Hat

DustinArmstrong, 2021-04-14:
Experiencing some screen tearing. It's not constant, but it happens often enough to say it's more often than not. I've uninstalled/reinstalled the software and hardware to ensure that it's unlikely to be the source of the issues.

Might anyone have some insight as to what could be causing this? It's always on the same axis

[image]

[image]

[image]

MikeGawi, 2021-04-14:
I had somethibg like that, on the same model, when used HAT and power supply that provides less than 3A. Try with the different PSU.

DustinArmstrong, 2021-04-14:
@MikeGawi - That was the issue. Plugged into a bigger power source (which I had to steal from another component) and that resolved the tearing. Thanks!

closed 2021-04-14 by DustinArmstrong