waveshareteam/e-Paper #155: Missing Documentation for Command 0xF8 (boost?)

txoof, 2021-04-14:
The [spec pdfs on the wiki](https://www.waveshare.com/w/upload/2/2d/2.7inch-e-paper-Specification.pdf) show `Power optimization` in the init flow chart and many of the [2in7 drivers](https://github.com/waveshare/e-Paper/blob/af9694c602fb24d776cd0f0b3e073a477294e3d8/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in7.py#L311) all use this line, but it is not documented clearly anywhere.

What does this feature do and what are the valid commands for it?

SSYYL, 2021-04-16:
I have looked up some information but there is no introduction about this register. I'll try to ask the factory.

SSYYL, 2021-05-07:
I asked the factory, they said they did not have relevant information of this register.
I think it should be Boost. It is directly provided by the IC manufacturer, and it is suggested not to modify it.

txoof, 2021-05-08:
@SSYYL 
Thanks for the follow up. Do you mean to say this is a required part of the startup procedure and should be used exactly as shown in the 2in7 drivers?

SSYYL, 2021-05-08:
Yes, but it might work without it. I haven't verified it.
This screen is relatively old, I have not seen similar registers on other screens.