waveshareteam/e-Paper #158: Waveshare 7.5inch HD (17087) not working with Arduino Shield (15082)?

FelixWeichselgartner, 2021-04-26:
Hi,

I have a 17087 Display 7.5 inch HD display wired up to an arduino shield 15082. Somehow I can't get output anything to the display. I tried the following sketches:
* https://github.com/waveshare/e-Paper/tree/master/Arduino/epd7in5_HD
* https://github.com/waveshare/e-Paper/tree/master/Arduino/epd7in5b_HD
* https://www.waveshare.com/wiki/E-Paper_Shield (epd7in5_HD & epd7in5b_HD from demo code).

[image]

The pixels seems to take on some noise state, like seen in this picture. Besides that nothing ever happens/happened.

[image]

Am I generally doing something wrong? Those parts should be compatible right? Can anyone confirm this behaviour? TY in advance.

SSYYL, 2021-05-13:
You didn't do anything wrong. I have no problem testing the program.(E-Paper_Shield)

FelixWeichselgartner, 2021-05-13:
@SSYYL TY very much for your response. I suppose I just have a defective part then.

Ediolot, 2021-05-15:
@FelixWeichselgartner It might not be this, but I initially tried with an Arduino UNO and it was not working at all. I thought it was defective part too, but I tried today with a Raspberry Pi A+ and it worked but I was having this exact issue: #154 Turns out their solution was correct, using a 3A PSU on the Raspberry and it works perfectly fine now. 

Maybe it's worth to check if the power source for the Arduino that you are using is not capable of giving 3A and try with one that does. If it's connected to the computer USB the maximum might be 0.5A.

(Also, the Raspberry Vout is 3.3V not the same as the 5V of the Arduino, but if the e-Paper hat has a regulator it should be ok)

FelixWeichselgartner, 2021-05-15:
@Ediolot ty for the tip. Indeed I was using my computer usb as power source. I will try with my lab bench power supply and reply result later.

FelixWeichselgartner, 2021-05-28:
@Ediolot : Your tip with the power supply didn't do the trick for me. But still thanks for your response.

I ended up purchasing this combo on Amazon (https://www.amazon.de/gp/product/B07DKJ46PQ/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) - 264x176 Resolution 2.7 Inch e-Paper Display HAT E-Ink Screen LCD Module SPI Interface with ESP8266 WiFi Driver Board.

This combo worked fine for me. Also I could connect the 17087 Display 7.5 inch HD to it and the demo (https://www.waveshare.com/wiki/E-Paper_ESP8266_Driver_Board) was running fine, like seen in the img.

[image]

I suspect it just a problem with either my Arduino Nano board or with the driver for it.