waveshareteam/e-Paper #159: Missing documentation for Data Start Transmission 2 (DTM2)

Novakov, 2021-05-06:
In specification for 4'2 display (linked here https://www.waveshare.com/wiki/4.2inch_e-Paper_Module, specification dated 2017.03.01, rev 2.0) there is description for command 0x10 (Data Start Transmission 1), however there is no documentation for command 0x13 (it seems to be Data Start Transmission 2). This command is used in driver code and mentioned in flowcharts.

SSYYL, 2021-05-07:
Yes, there are indeed gaps in this specification, I hvae uploaded an IC specification:[UC8176](https://www.waveshare.net/w/upload/8/88/UC8176.pdf)

Novakov, 2021-05-07:
Thanks! Could you also add link to this datasheet on wiki page? It would be much easier for anyone to find it?

SSYYL, 2021-05-08:
Of course, I will upload to wiki page.

closed 2021-07-01 by Novakov