waveshareteam/e-Paper #166: 7in5_V2 bitmap gets BW inverted using C script

uros76, 2021-05-13:
Hi. I have this screen V2 https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT
When I try to send an 800x480 image to the screen it works using python script. 

But, when I try to do the same with C script, the image always gets inverted. The black is white and the white becomes black on the screen. I tried different things on C script and every time the same problem. Is there some trick to fix this and get the image on the screen as original?

SSYYL, 2021-05-14:
In function: `void EPD_7IN5_V2_Display(const UBYTE *blackimage)`
Invert when sending data.
modification
`EPD_SendData(~blackimage[i + j * Width]);`
to
`EPD_SendData(blackimage[i + j * Width]);`

uros76, 2021-05-14:
Sorry, I don't fully understand what I need to do :(  

In your example script, on which line should I change what? 
https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/c/examples/EPD_7in5_V2_test.c

missionfloyd, 2021-05-14:
Line [206](https://github.com/waveshare/e-Paper/blob/83a6cffae79c83ef49724cfe6dd58e22245009cc/RaspberryPi_JetsonNano/c/lib/e-Paper/EPD_7in5_V2.c#L206) in `lib/e-Paper/EPD_7in5_V2.c`. Remove the `~`

SSYYL, 2021-05-17:
https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/c/lib/e-Paper/EPD_7in5_V2.c

line 206 :)

uros76, 2021-05-17:
That did the trick. Thanks a lot for help @SSYYL !!!

closed 2021-05-17 by uros76