waveshareteam/e-Paper #171: Python Examples - INFO:root:[Errno 2] No such file or directory

PolarisNeo, 2021-06-23:
I recently got a Waveshare 2.3" e-Paper HAT. I have followed through all of the setup guides and crawled through the documentation. Unfortunately, whenever I try to run any of the Python examples I am faced with this error: "INFO:root:[Errno 2] No such file or directory".

I tried the C examples too, these also failed - although my C knowledge is pretty much non-existent.

SPI is enabled, I am pretty sure the module is not damaged it came in pristine condition and all I have done is put it on the RPI Zero pins, I have checked the connections and they should all be fine. 

Any ideas?

PolarisNeo, 2021-06-24:
I've narrowed it down to find it is failing on this particular line in epd_2in13_V2_test.py
`    epd.init(epd.FULL_UPDATE)
`

In more detail:
`>>> epd.init(epd.FULL_UPDATE)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/waveshare_epd-0.0.0-py3.7.egg/waveshare_epd/epd2in13_V2.py", line 124, in init
    if (epdconfig.module_init() != 0):
  File "/usr/local/lib/python3.7/dist-packages/waveshare_epd-0.0.0-py3.7.egg/waveshare_epd/epdconfig.py", line 74, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory
`

Any ideas on how to resolve this? I'm using the 2.3" e-Paper HAT attached directly using the hat on the GPIO pins of a Raspberry Pi ZeroW.

I think something is wrong with the SPI, I've checked and it's all enabled - or if it is enabled it might not be working but I don't know where to go from here. 

Any help appreciated

SSYYL, 2021-06-26:
Looks like an SPI problem. 
Try：` sudo pip3 install spidev` or `sudo pip install spidev`

kpmpower, 2021-07-23:
Have you tried what was recommended at [this issue](https://github.com/waveshare/e-Paper/issues/5#issuecomment-479632988). It worked for me after bashing my head for a while. The key for me was `sudo rpi-update`

PolarisNeo, 2021-07-29:
Thanks for the suggestions guys, unfortunately those didn't work for me. However, I completely wiped my RPI-Zero with a fresh install and everything is working ok now. My RaspiOS must've just been corrupted even though it looked fine.

closed 2021-07-29 by PolarisNeo

kushalarya, 2021-12-25:
I debugged this, this error was coming from file `python/lib/waveshare_epd/epdconfig.py` on line `self.SPI.open(0,0)` in function `module_init`. 

Enabled the SPI, it worked.

1. Run `sudo raspi-config`
2. Select  `Interfacing Options`
3. Arrow down to `SPI`
4. Select `yes` when it asks you to enable `SPI`

Marc2629, 2023-09-05:
@kushalarya Thanks so much, this is missing from the official guide.

aaronperkel, 2024-08-18:
I am having this same issue, I have tried enabling SPI and I can not get it to work