waveshareteam/e-Paper #174: Publish Python library on PyPi

arseniiarsenii, 2021-07-13:
Our project uses this Python library for the [2.13inch e-Paper HAT (C)](https://www.waveshare.com/product/displays/e-paper/2.13inch-e-paper-hat-c.htm), however there is no official package on PyPi, so dependency management and updating is complicated since we have to manually place required files in our project directory.

There are some unofficial PyPi packages for this library, but they do not seem to be properly maintained and are not based on the official repo, but rather on its forks. Thus, those packages are not suitable for the production environment.

txoof, 2021-07-16:
You can try:
`pip install -e 'git+https://github.com/waveshare/e-Paper.git#egg=waveshare-epd&subdirectory=RaspberryPi_JetsonNano/python'`

and in your `requirements.txt` use `git+https://user@bitbucket.org/repo-name/scripts.git#subdirectory=folder1/module`

I was running into problems with this and PyInstaller and just gave up trying to diagnose it instead I use the script below to clone and extract just what I need.

`refresh_waveshare.sh`
```
!/bin/bash
# waveshare repo
ws_epd_git="https://github.com/waveshare/e-Paper.git"

ws_root="e-Paper"
ws_library_path="e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd"

# project waveshare_epd path
project="./paperpi/"
ws_local="$project/waveshare_epd"

# create temporary directory
ws_tmp=$(mktemp -d -t waveshare_XXXXX)
pushd $ws_tmp
git clone $ws_epd_git
if [ $? -ne 0 ]; then
  echo "failed to clone $ws_epd_git"
  echo "see $ws_tmp"
  echo "exiting"
  exit 1
else
  pushd $ws_root
fi

echo ws_temp = $ws_tmp

# get the latest commit and store it
ws_version=$(git log -1 --format=%h\ %ci)
echo "this version is: $ws_version"
popd
popd

# backup the current version of the library
rm -r $ws_local.ignore
mv $ws_local $ws_local.ignore
cp -r $ws_tmp/$ws_library_path $project
# add the latest commit to the constants file for record keeping (?)
sed -i "s#\(ws_version\s\?=\).*#\1 '$ws_version'#g" $project/constants.py

echo "cleaning up temporary directories"
#rm -rf $ws_temp

exit 0
```

arseniiarsenii, 2021-07-16:
Unfortunately I cannot apply this solution, since our project uses Poetry for dependency management, which has very limited support for git module installation.

arseniiarsenii, 2021-07-16:
Wouldn't it be easier to just publish the library on PyPi and configure Git actions to update it on push rather than coming up with some complicated workarounds?

txoof, 2021-07-21:
The script included above might be an alternative. Just consider the WaveShare stuff static data and add it manually, or as part of a build script.

Based on the multiple unfulfilled requests for a PyPi module in the historical issues, there probably aren't many other options.

SSYYL, 2021-07-23:
@arseniiarsenii Hi, If you only use one model screen, I think the method of @txoof  can be adopted completely.
by the way,  2.13inch e-Paper HAT (C) are almost out of production.

txoof, 2021-07-25:
@arseniiarsenii 

You may be interested in the epdlib I'm working on for your project as well. When implented, it makes it easier to use most of Waveshare's displays with your project. It helps avoid locking users into one particular display.

Rather than hard-coding your display output for one resolution, it allows you to provide scalable layouts that will work on almost any EPD. 

https://github.com/txoof/epdlib

It's still in development, but the API is stable and I'll be pushing a new version when I return from holiday.