waveshareteam/e-Paper #178: Raspberry Pi Zero W + 7.5inch E-Paper HAT python code displays nothing

andykais, 2021-07-16:
Hi, I have a 7.5inch e-paper display that was bought in June 2021. Running the example python code displays nothing on the screen. The c code correctly displays the example images on the screen.

<details>
  <summary>python <code>epd_7in5_V2_test.py</code> output</summary>
<pre>
(e-paper) root@raspberrypi:/home/pi/Code/scratchwork/waveshare-eink-libraries/e-Paper/RaspberryPi_JetsonNano/python# python examples/epd_7in5_V2_test.py 
INFO:root:epd7in5_V2 Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:1.Drawing on the Horizontal image...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:2.Drawing on the Vertical image...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:3.read bmp file
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:4.read bmp file on window
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Clear...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:Goto Sleep...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...
(e-paper) root@raspberrypi:/home/pi/Code/scratchwork/waveshare-eink-libraries/e-Paper/RaspberryPi_JetsonNano/python# exit
</pre>
</details>

<details>
  <summary>c code<code>./epd</code>output</summary>
<pre>
(e-paper) pi@raspberrypi:~/Code/scratchwork/waveshare-eink-libraries/e-Paper/RaspberryPi_JetsonNano/c $ sudo ./epd 
EPD_7IN5_V2_test Demo
/***********************************/ 
Current environment: Raspbian
bcm2835 init success !!! 
/***********************************/ 
e-Paper Init and Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
5 S
Paint_NewImage
show window BMP-----------------
pixel = 800 * 480
Debug: e-Paper busy
Debug: e-Paper busy release
show bmp------------------------
pixel = 100 * 100
Debug: e-Paper busy
Debug: e-Paper busy release
show image for array
Debug: e-Paper busy
Debug: e-Paper busy release
SelectImage:BlackImage
Drawing:BlackImage
EPD_Display
Debug: e-Paper busy
Debug: e-Paper busy release
Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Goto Sleep...
Debug: e-Paper busy
Debug: e-Paper busy release
close 5V, Module enters 0 power consumption ...
</pre>
</details>

SSYYL, 2021-07-19:
There's nothing wrong with the code, One thing to note is that after running the C code (library BCM2835), you need to restart the device to run python code

Edge720, 2021-07-21:
Hi,

I've also been having this issue when running epd_7in5_V2_test.py with a display and HAT bought July 2021.

SSYYL, 2021-09-24:
Hello, I did find this problem recently, which may be caused by the long reset time of low level.
Try to reduce the low level delay time (<2ms) in reset(self).

evanthomas, 2021-11-15:
Yep, also having this problem. How do I "restart the device"?