waveshareteam/e-Paper #188: 2in9 showing only odd mosaic, not responding for example code

unster, 2021-08-15:
Hi team. 

After sucessful connection to RP Zero, I've tried to test the screen with epd_2in9_test.py - but the screen shown nothing, and with epd_2in9_V2_test.py I've only got this - https://i.ibb.co/Nn8DnSX/IMG-20210815-225054.jpg.

Next tries did not change to what is being displayed on screen.

Any ideas what is the issue?

SSYYL, 2021-08-18:
Hi @unster,

I tested it with RPi Zero and it worked. 
I suggest you contact our after-sales staff for a better analysis.

PhilippLuebeck, 2021-08-18:
I had the same issue with my 4.2" display at the beginning. I then had realized that I had forgotten to initialize the DC pin properly on my 32bit MCU and after configuring the DC pin to PushPull mode the display worked perfectly. But I am not sure if this is the case on your RP Zero as well.

jaredmeyers, 2021-09-06:
@unster I am having a similar issue with the 2.13 inch. It just shows a static like pattern and doesn't respond to the example code. I'm also using a RPi ZeroW.
Have you found a fix yet?

betasmiths-stan, 2023-11-14:
a note for the python code: I was using RPi 4b and had the same issue - turns out epdconfig.py uses `RPi.GPIO.setmode(RPi.GPIO.BCM)` instead of RPi.GPIO.BOARD. Since there are differences between difference version of raspberry BCM pins and board pins, that caused the mosaic for me. I just redefined the pins to boardpins and updated the module_init to use `self.GPIO.setmode(self.GPIO.BOARD)` and everything worked