waveshareteam/e-Paper #190: self.send_command(0x71) command in def ReadBusy for epd7in5_V2.py

tkameroski, 2021-08-20:
In the file epd7in5_V2.py, the ReadBusy definition contains the command self.send_command(0x71) twice.  Once at the start (line 132) of the def then again in the while loop (line 135).  While trying to figure out why my Pi zero W occasionally enters the while loop but never exits it, I noticed that if I comment out both instances of self.send_command(0x71), the driver seems to work normally.  What does that command do?  It doesn't appear to be necessary if the driver works normally without it.  BTW, removing them doesn't solve the endless loop problem that occurs every so often.  I noticed that the drivers for many other epd displays do not include those lines.  Thanks.

   def ReadBusy(self):
        logger.debug("e-Paper busy")
        self.send_command(0x71)
        busy = epdconfig.digital_read(self.busy_pin)
        while(busy == 0):
              self.send_command(0x71)
              busy = epdconfig.digital_read(self.busy_pin)
        epdconfig.delay_ms(20)
        logger.debug("e-Paper busy release")

SSYYL, 2021-08-28:
0x71 is left over from the previous version and has no effect now.
Try using an external 8-pin cable. If possible, please give me a clear picture of the back.

tkameroski, 2021-08-28:
I have been using the 8-pin cable.
[image]

closed 2021-09-03 by tkameroski