waveshareteam/e-Paper #191: Cant exit busy state.

Yasen6275, 2021-08-28:
Hello,

I have 2.13 inch tricolor(RBW) pi hat. When it arived I tested it and it was working ok. After that I postponed the project and had to reuse the sd card pi for something else. When I tried it again on pi0W I wasn't able to get out the screen from busy state  neither with c program nor with python script.  Pi is fully updated. I tried newest bcm lib and got segfolths. I reformatted the SD card and used 1.60 version like it is advised here: https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT_(B).
`# python3 examples/epd_2in13bc_test.py 
INFO:root:epd2in13bc Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd2in13bc:e-Paper busy`
`# ./epd 
EPD_2IN13BC_test Demo
/***********************************/ 
Current environment: Raspbian
bcm2835 init success !!! 
/***********************************/ 
e-Paper Init and Clear...
Debug: e-Paper busy` 
Any idea how to debug/fix the problem?

Yasen6275, 2021-08-29:
I have learned some procedure programing long time ago on pascal, so I might be wrong but...

`void EPD_2IN13BC_ReadBusy(void)
{
    Debug("e-Paper busy\r\n");
    while(DEV_Digital_Read(EPD_BUSY_PIN) == 0) {
        DEV_Delay_ms(100);
    }
    Debug("e-Paper busy release\r\n");
}`
What will trip busy pin to low? Isn't that an endless loop?

Yasen6275, 2021-08-29:
Same result with fully upgraded raspbian strech.

Yasen6275, 2021-08-29:
Tried recompiling example against wiringpi libs. Example run without image and without any errors. Reverted back to bcm libs, and it didin't stuck but no image ether.

tkameroski, 2021-08-29:
I had the same or very similar problem with a 7.5 inch display when using the python version of the driver.  A closed issue thread https://github.com/waveshare/e-Paper/issues/30 contained a work-around that worked for me.  It replaces some of the lines of lines of code in the ReadReady loop you described above to break the loop if its try count goes over 150.  Maybe a similar code would work for the c driver if that version of the driver is your problem

Here is the code I used to replace the def ReadBusy(self) block of the python code;  I used .. to indicate spaces because the spaces disappear when I submit the comment.

.....logging.debug("e-Paper busy")
.....iter = 0
.....while epdconfig.digital_read(self.busy_pin) == 0:
..........epdconfig.delay_ms(100)
..........iter += 1
..........if iter > 150:
...............logging.info("Forced e-paper busy release")
...............break
.....logging.debug("e-paper busy release")

cvasquez-github, 2022-05-26:
Could it be related to the SPI (CLK and DIN) pins not connected correctly?
For Arduino UNO you should use D13 and D11 like on WaveShare Wiki, but make sure you use the right pins according to your board. For Arduino MEGA 2560 for instance should be D51 and D52. https://github.com/cvasquez-github/arduino-mega-epaper