waveshareteam/e-Paper #194: 2.13 inch HAT showing static pattern and not responding to example code

jaredmeyers, 2021-09-06:
I am using a Raspberry Pi ZeroW and am trying to use the Python3 example code to work with my display. It just shows a static pattern and doesn't respond at all when I run the example code from this repo. I've installed all the dependencies it says to and checked to make sure all the pins works. 
How do I fix this?

SSYYL, 2021-09-24:
Hello, I did find this problem recently, which may be caused by the long reset time of low level.
Try to reduce the low level delay time (<2ms) in reset(self).
In addition, may I ask whether the C program can be used normally?

jaredmeyers, 2021-09-24:
@SSYYL the C programs don't work either. It may just be a hardware connection issue.

SSYYL, 2021-09-24:
It was the hardware design that caused some of the problems.
Try further shortening the low level time by, like 0.2ms.

Inviz, 2021-09-28:
I found a bug https://github.com/waveshare/e-Paper/issues/198 maybe this is related. Width was set to 128, not 122

jaredmeyers, 2021-09-28:
But I primarily tried the python code and that didn't work either. It's not
limited to C.

Inviz, 2021-09-28:
Sorry to hear that. two of the units i tried work on raspberry with the provided py code. I used 2.13 touch version.

jaredmeyers, 2021-09-28:
Another screen (not the same company or type) worked after I re soldered,
so I think it may be a connection issue. I'll try this screen again on next
Pi project.