waveshareteam/e-Paper #207: Can't display black with demo on 4in2b

dubiao, 2021-11-27:
https://github.com/waveshare/e-Paper/blob/bcb561fd70f146e733d3e2555254dfa1bd7ad4c8/Arduino/epd4in2b_V2/epd4in2b_V2.cpp#L211

I am using 4in2b to show demo but black image alway been ignored. 

I change ClearFrame() function to following:
```c
void Epd::ClearFrame(void) {
    SendCommand(DATA_START_TRANSMISSION_1);           
    DelayMs(2);
    for(int i = 0; i < width / 8 * height; i++) {
        SendData(0xF0);     // it was 0xff
    }  
    DelayMs(2);
    SendCommand(DATA_START_TRANSMISSION_2);           
    DelayMs(2);
    for(int i = 0; i < width / 8 * height; i++) {
        SendData(0x0F);   // it was 0xff
    }  
    DelayMs(2);
}
```
It should display black and red vertical lines but only red lines are shown.. 

I also tried other codes but always showing red but none black. 
Any advice?

dubiao, 2021-11-27:
And .
If I try to display the demo image with pass in params with black and red to show this:
我还试了Demo中显示如下图片的代码：
[image]

but this is what result to:
但只得到了图中红色显示的图案。
[image]

Seems that event the red image' bits are reverted. 
而且颜色还反转了？？那个部分应该是白底红图，但显示了红底白图。

What happened to this? I use different tow screen (both 4in2 red black white), but the same result.
我试了两块红黑白三色屏幕，4.2寸的，但效果一样。应该不是屏幕的问题吧？

SSYYL, 2021-12-14:
你好，你的理论预期是对的，直接运行我们示例程序也应该是有黑白红三色显示的，如果是购买的我司的屏幕你可以联系我们销售进行售后服务。
如果不是我们的产品我推测可能是某些寄存器默认值不同，而我们程序没有重新写值直接使用的OTP设置导致的