waveshareteam/e-Paper #208: text rotate 90 or 270 degree

michapr, 2021-12-04:
Hi,

I can't get to work the text rotation to 90 or 270 degree.
```
...
paint.Clear(COLORED);
paint.SetRotate(1);
paint.DrawStringAt(20, 5, "Hello world!", &Font16, UNCOLORED);
epd.TransmitPartialData(paint.GetImage(), 0, 64, paint.GetWidth(), paint.GetHeight());
...
```
What is wrong here? Already tried several things, but no success...
(using 2.7 inch display)

Thanks!
Michael

SSYYL, 2021-12-14:
Hi, When rotating text, you need to modify the frame buffer slightly, otherwise the display will not be complete.
```
  unsigned char image[1024];
  Paint paint(image, 24, 176);    //width should be the multiple of 8
  paint.SetRotate(1);
  
  paint.Clear(UNCOLORED);
  paint.DrawStringAt(5, 5, "e-Paper Demo", &Font16, COLORED);
  epd.TransmitPartialData(paint.GetImage(), 48, 0, paint.GetWidth(), paint.GetHeight());
  
  paint.Clear(COLORED);
  paint.DrawStringAt(5, 5, "Hello world!", &Font16, UNCOLORED);
  epd.TransmitPartialData(paint.GetImage(), 0, 0, paint.GetWidth(), paint.GetHeight());
```
hope this helps, good luck.

glogge64, 2022-06-24:
@SSYYL

I have the e-paper version epd_1in54_V2 an also can't rotate 270 degrees.
How do I have to modify the frame buffer?

Below I have attached both codes for no rotation and 270 degree rotation.
I hope you can help me.

**0 degree:**
```
#include <SPI.h>
#include "epd1in54_V2.h"
#include "imagedata.h"
#include "epdpaint.h"
#include <stdio.h>

Epd epd;
unsigned char image[1024];
Paint paint(image, 0, 0);

#define COLORED     0 // Black
#define UNCOLORED   1 // White

void setup()
{
  Serial.begin(9600);

  // epd.LDirInit(); // low init --> black
  epd.HDirInit(); // high init --> white
  epd.Clear();

  paint.SetWidth(200); // x-direction
  paint.SetHeight(15); // y-direction

  paint.Clear(COLORED);
  paint.DrawStringAt(0, 0, "Overview", &Font16, UNCOLORED);
  epd.SetFrameMemory(paint.GetImage(), 0, 185, paint.GetWidth(), paint.GetHeight());
 
  epd.DisplayFrame();
}
```
**270 degree:**
```
#include <SPI.h>
#include "epd1in54_V2.h"
#include "imagedata.h"
#include "epdpaint.h"
#include <stdio.h>

Epd epd;
unsigned char image[1024];
Paint paint(image, 0, 0);

#define COLORED     0 // Black
#define UNCOLORED   1 // White

void setup()
{
  Serial.begin(9600);

  // epd.LDirInit(); // low init --> black
  epd.HDirInit(); // high init --> white
  epd.Clear();

  paint.SetRotate(3); // 1 = 90°; 2 = 180°; 3 = 270°

  paint.SetWidth(200); // x-direction
  paint.SetHeight(15); // y-direction

  paint.Clear(COLORED);
  paint.DrawStringAt(0, 0, "Overview", &Font16, UNCOLORED);
  epd.SetFrameMemory(paint.GetImage(), 0, 185, paint.GetWidth(), paint.GetHeight());
 
  epd.DisplayFrame();
}
```