waveshareteam/e-Paper #211: Support for 4in3 UART ePaper?

giddyhup, 2021-12-15:
Do you plan to provide a python driver for your 4.3 inch UART ePaper? In particular an example on how to transmit image data to the screen module from a Raspberry Pi would be highly appreciated. Thanks!

SSYYL, 2021-12-17:
We plan, but there's too much planning,
Several C versions are available on our wiki, And there's a user's manual.

Rewrite into Python development ideas is UART sending and receiving, 
you can refer to our other RPi UART module programming.

good luck.

giddyhup, 2021-12-21:
@SSYYL, thank you for your reply. Regrettably, my C skills do not reach very far but still, could you please point be to the example you mentioned? I do not see one for your 4.3 inch ePaper. There are examples for a 4in37b display but the code mentions the color red so I think it is not meant for the four-gray 4.3" UART type.

SSYYL, 2021-12-22:
@giddyhup , Information can be found on this page: [Waveshare Wiki](https://www.waveshare.com/wiki/4.3inch_e-Paper_UART_Module)

The 4.3" UART is not the same type of screen as the one mentioned in this repo.

giddyhup, 2021-12-22:
@SSYYL I was able to extend the Python module I found [here](https://github.com/jarret/raspi-uart-waveshare/blob/504207888f5fee8509cf2494640319afb6977de3/waveshare/epaper.py). It appears I can initiate a transfer. 

I send the proper command sequence and I get some positive response from the module but the file does not get saved. The module acknowledges the transfer: `OKSave the file as: TEST.BMP`. But even though I send a 'y' as per the instructions (`self.serial.write(b'y\r'\n)`) the file does not get saved to the SD card.

I think I am close to solving it. What am I missing?

giddyhup, 2021-12-23:
I think I am very close to sorting it out. Here are some more details: After sending the sequence for the upload I get the response: `OKSave the file as: TEST.BMP\r\nPlease send a file (less than 256KB) using UART...`. I then transmit the file. At the end the module reports the file size: `File size: 20920\r\n`. I then send `y\r\n`.  

Still, the file doesn't get written to the card. My serial write command returns 20914 bytes written (the same size as reported the by the operating system). I verified that the file can be displayed by manually writing it to the card.

giddyhup, 2021-12-23:
Last remark. I figured it out and extended an existing Python library. The [repository can be found here](https://github.com/giddyhup/raspi-uart-waveshare).

closed 2021-12-23 by giddyhup

SSYYL, 2021-12-24:
Great work and congratulations!