waveshareteam/e-Paper #212: partial refresh 3.7" display - Arduino

michapr, 2021-12-16:
Hi,

can you provide a sample how to make a partial update/refresh?
As sample for weather station, where is only changing the temperature - I need to make update only at that place of display.

If I use
`DisplayFrame_Partial(const UBYTE *Image, UWORD Xstart, UWORD Ystart, UWORD iwidth, UWORD iheight)`
the content of *Image will be shown, but the rest of the display will be cleared.

I cannot get it, how to **_*update*_** a part of the screen only without to clear existing content.

Thanks!
Michael

SSYYL, 2021-12-29:
An example has been added: #215

michapr, 2021-12-29:
Thanks!
With modified library and LUT it is working now in partial refresh.

That the old content in background is still visible (in light grey) is normal?
Or can this be optimized by any parameter?
I understand that this cannot be a full init, but I have seen at other displays a partial refresh with less visible old content.
Any idea?

Thanks!
Michael

michapr, 2021-12-29:
Thanks!
With modified library and LUT it is working now in partial refresh with your sample.

That the old content in background is still visible (in light grey) is normal?
Or can this be optimized by any parameter?
I understand that this cannot be a full init, but I have seen at other displays a partial refresh with less visible old content.
Any idea?

In your sample it is working fine with background image, but how to realize with strings as sample at different places?
-------------------
| string1 | string2 |
-------------------
| string3 | string4 |
-------------------

In this table I cannot update string2 (or any other place) without flickering of the rest of the screen.
Can you add in your sample a second string - and update at first the time string - and then (as sample) print out the counter at other place in other string?

BTW: A partial "full refresh" (init) is not possible? So that as sample from time to time such a full refresh could be done?

Thanks!
Michael

SSYYL, 2021-12-31:
Use GC to clean up the gray
`epd.DisplayFrame_Part(... , isGC = true)`

Two pieces of RAM display alternately, Old data must be written along with new data.

michapr, 2021-12-31:
>epd.DisplayFrame_Part(... , isGC = true)

Have tried, but see no difference. What should happen here?

>Two pieces of RAM display alternately, Old data must be written along with new data.

Do not understand, what do you mean?
Partial refresh in table (as sample) will not work? I should write whole table?

If I have a graph at the top and want to refresh only values in table - I should write the full graph every time again? Cannot believe it.

Maybe I do not understand right how to make a partial refresh with elements created with paint.Draw... 
Should I create different paint elements (paint1, paint2,...) and change their content? But how to refresh only this changed content of one element on display? 
If using Epd::DisplayFrame_Part() - can not be refreshed ONLY this part? 
I thought this is the sense of *partial* refresh support?

Maybe I miss something in my understanding - thank you for your help!

SSYYL, 2022-01-10:
You don't have to write the entire table, You only need to write the last modified data again after refresh and then write the data you want to update.
For details, see #221

michapr, 2022-01-12:
THANKS! 

Now it is working as I  expected !

Thank you for your support!
Michael

closed 2022-01-12 by michapr

winstonma, 2022-01-16:
Hi

I am trying to add this feature on my 7.5" eink display, using RPi. Is it possible to add partial refresh on all example? Thanks