waveshareteam/e-Paper #230: e-Paper and old Raspberry Pi B V2.0 -- C code works, but python doesn't

timc995, 2022-02-28:
I have a 7x5 v2 board that works great with my Pi 4.  I tried to use it with a much older Pi B V2.0. 

What works:
- The C test code here (epd) works fine when I configure it with USELIB_RPI=USE_BCM2835_LIB
This confirms that I have attached the device propertly to the Pi device

What does not work:
- The C test code here (epd) does not work when I configure it with USELIB_RPI=USE_WIRINGPI_LIB
- The python code does not work. 

In both of these cases, there is no error displayed, but the waveshare device is not updated.

dstubbs95, 2022-04-04:
@timc995 I am experiencing the same issue. I am using a RPi 2B and the c-code works perfectly fine with the BCM2835 library, whereas the python version doesn't yield any response from the screen. I don't get any errors from running the python code either. Any luck finding a solution to this issue?

timc995, 2022-04-04:
Hi Daniel.
Unfortunately I haven't got it to work.  You're the first to reply to me.
I think it must be an error with the waveshare library.  I looked for the
source code for the drivers but I didn't find it so I gave up.
Tim

dstubbs95, 2022-04-04:
> 

Tim,

I was able to get it working today. To be honest, I am not sure exactly what solved the issue. As part of my troubleshooting, I wired the MOSI to MISO pins on the RPi to create a loopback on the SPI interface. My first attempt was using the spidev-test program available at: [https://github.com/rm-hull/spidev-test](https://github.com/rm-hull/spidev-test). I got nothing but zeros back on the SPI bus, so I started to suspect that something was wrong with the spidev library. To further troubleshoot, I installed the "pigpiod" and "pigpio-tools" packages. With those installed you can run 
`sudo pigpiod`
`pigs spio 0 50000 0`
`pigs spix 0 1 2 3 4`
and you should get a response of "4 1 2 3 4". I got the correct response, which meant the SPI bus was working and further pointed to some issue with spidev. At that point I installed the "spi-tools" package to try and do some more troubleshooting with spidev. With the spi-tools package, you can use the `spi-pipe` program to do a similar loopback test as with pigpio as
`echo "test" | spi-pipe -d /dev/spidev0.0 | cat`
and you should see "test" echoed to your terminal. 

At this point spidev-test and the python version of the epd codes started to work as expected. I am not sure what part of this process fixed the issue. I thought that maybe one of the packages I installed brought in some needed dependency, but even after uninstalling those packages, things still seem to be working. I will say that the first time I tried the `spi-pipe` command, I got garbage back. Upon rebooting things seemed to be working. Hopefully some of this information can help someone else figure out their issue. If I get bored, I might try with a fresh install of Raspbian to see if I can't nail down exactly where the problem lies.

timc995, 2022-04-05:
I'm now motivated to start over.  I'm pretty sure I reused the SD card, but
I will spin another image up and see if I can replicate your success!  I'd
love to find a use for this very old Pi device, and driving an epaper
display would be great.

Lawlez, 2022-04-09:
@dstubbs95  Thank you, i followed all your steps as described, even tho i did not recieve the same output as you during the pigpio tests after all steps the default waveshare examples started to work on my raspi! 

so this worked: 
install pigpio and (re)start pigpiod > install pigpio-tools > install spi-tools > reboot > try examples again

1ou2, 2022-05-13:
I had the same issue on a Raspberry Pi Zero W Rev 1.1
The procedure described by @dstubbs95  and summarized by @Lawlez  worked perfectly for me.
Thanks !

Poohl, 2022-06-23:
Hey, I'm having the a similar issue on my Raspi 4B and found that a reboot reliably fixes it while, while the c code reliably causes it. Could someone verify that this is in fact the same problem?

Sequence of events to reproduce:
- start up raspi
- run `sudo python3 epd_XXXXX.py` -> screen updates
- run `sudo ./epd` -> screen updates
- run `sudo python3 epd_XXXXX.py` -> no response from screen
- run `sudo ./epd` -> screen still updates
- reboot
- run `sudo python3 epd_XXXXX.py` -> screen updates
- rinse and repeat.

bkahlert, 2023-07-20:
Had the same problem getting the Python demo located at `~/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in13_V3_test.py` to work.

But for some reason, the magic lines by @dstubbs95 got it working:
```shell
sudo apt-get install -y pigpiod pigpio-tools spi-tools
sudo pigpiod
pigs spio 0 50000 0
pigs spix 0 1 2 3 4
echo "test" | spi-pipe -d /dev/spidev0.0 | cat
```

Running the demo now works on my Raspberry Pi Zero W:
```shell
(cd ~/e-Paper/RaspberryPi_JetsonNano/python/examples && sudo python3 epd_2in13_V3_test.py)
```