waveshareteam/e-Paper #231: New e-paper lib produces dim output (Pi Zero W, revised hat, 7.5" display)

LLPP65, 2022-03-06:
My Pi Zero W and 7.5" display have been working flawlessly after the original hat was replaced with an updated one. Today (6 March 2022) I tried the current MASTER version of the e-paper library and this produces faded output on the display. After reverting to the old version of the library, the display output is correct (dark black).

I have the identical software on a Raspberry 3b with hat and the new library works fine. The display output is dark black and the refresh time is faster than the old library.

beniaminogreen, 2022-03-07:
Thanks for leaving this comment - I had the same issue setting up my display. How far back did you have to go to get clear images?

Now all I have to do is fix the display freezing problem...

LLPP65, 2022-03-07:
The version of epd7in5_V2 that I have working with the Pi Zero W has these lines in the header:

# * | This version:   V4.0
# * | Date        :   2019-06-20
# # | Info        :   python demo
#
# Modifications by Dmitry Brant:
# - Oct 2020: New routines for faster SPI data transfer.

beniaminogreen, 2022-03-07:
Super, thanks so much!