waveshareteam/e-Paper #233: epd2in13_V2 shows everything mirrored in x axis

JohnJonnyJohn, 2022-04-03:
I'm currently using the example code of [epd2in13_V2](https://github.com/waveshare/e-Paper/tree/master/Arduino/epd2in13_V2), but everything is mirrored within the x axis. Im pretty new into this kind of stuff, because of that Im not sure if there is an issue or Im just doing it wrong.

This is my code:

```
#include <SPI.h>
#include "epd2in13_V2.h"
#include "epdpaint.h"
#include "imagedata.h"

#define COLORED     0
#define UNCOLORED   1

unsigned char image[4000];
Paint paint(image, EPD_WIDTH, EPD_HEIGHT);
Epd epd;

void setup()
{
  epd.Init(FULL);
  epd.Clear();
  
  paint.Clear(UNCOLORED);
  
  paint.DrawStringAt(10, 100, "Hello world!", &Font12, COLORED);
  paint.DrawHorizontalLine(10, 10 , 100, COLORED);
  
  epd.Display(paint.GetImage());

  epd.Sleep();
}
```
Problem: 
[image]

alexei-YUV, 2022-11-02:
Hi John, 

I solved by this issue by modifying epd2in13_V2.cpp file.
Inside this file search for:
 **void Epd::Display(const unsigned char* frame_buffer)**

and switch the line: **for (int j = 0; j < h; j++) {**
with **for (int j = h; j >= 0; j--) {**

Cheers,

tiannahaha, 2023-05-05:
hello , I've met the same mirroring problem. 
I tried to modify the code in` EPD_7in5b_V2.cpp`, but the variable` j `is of type` UDOUBLE `in the for loop.
That is to say,`j` is always greater than 0, for expressions with `j>=0`
I used the code `for (int32_t  j = Width; j >= 0; j--)`, it's not work.
How can I fix it?

alexei-YUV, 2023-05-05:
> hello , I've met the same mirroring problem. I tried to modify the code in` EPD_7in5b_V2.cpp`, but the variable`j`is of type`UDOUBLE`in the for loop. That is to say,`j` is always greater than 0, for [...]

Hi, 
The function to show the image on bigger display is more complex but should apply the same method to mirror again by starting from another corner. Please try to replace in function "Epd::Displaypart" the following for loops:

`for (unsigned long j = 0; j < height; j++) {
    for (long i = width/8 - 1; i >= 0; i--) {
        if( (j>=yStart) && (j<yStart+Picture_Height) && ((width - (i*8+8))>=xStart) && ((width - i*8)<=xStart+Picture_Width)){
			
            if(Block == 0){
                SendData((pgm_read_byte(&(pbuffer[(width/8-1-i)-xStart/8 + (Picture_Width)/8*(j-yStart)]))) );
            }else{
                SendData(~(pgm_read_byte(&(pbuffer[(width/8-1-i)-xStart/8 + (Picture_Width)/8*(j-yStart)]))) );
            }
			
        }else {
          if(Block == 0){
            SendData(0xff);
          }else {
            SendData(0x00);
          }
        }
    }
}
`

Let me know if this works,

alexei-YUV, 2023-05-05:
> > Hi John,
> > I solved this issue by modifying epd2in13_V2.cpp file. Inside this file search for: __void Epd::Display(const unsigned char_ frame_buffer)_*
> > and switch the line: **for (int j = 0; j < h; j++) {** with **for (int j = h; j >= 0; j--) {**
> > Cheers,
> 
> hello , I've met the same mirroring problem. I tried to modify the code in` EPD_7in5b_V2.cpp`, but the variable`j`is of type`UDOUBLE`in the for loop. That is to say,`j` is always greater than 0, for expressions with `j>=0` I used the code `for (int32_t j = Width; j >= 0; j--)`, it's not work. How can I fix it?

Hi, 

In function Epd::DisplayPart try to modify these lines(line 138, 139, 141 and 143) with:
`for (long i = width/8 - 1; i >= 0; i--) {`
`if( (j>=yStart) && (j<yStart+Picture_Height) && ((width - (i*8+8))>=xStart) && ((width - i*8)<=xStart+Picture_Width)){`
`SendData((pgm_read_byte(&(pbuffer[(width/8-1-i)-xStart/8 + (Picture_Width)/8*(j-yStart)]))) );`
`SendData(~(pgm_read_byte(&(pbuffer[(width/8-1-i)-xStart/8 + (Picture_Width)/8*(j-yStart)]))) );`

Let me know if this works,

tiannahaha, 2023-05-05:
> Hi,
>
> In function Epd::DisplayPart try to modify these lines(line 138, 139, 141 and 143) with: `for (long i = width/8 - 1; i >= 0; i--) {` `if( (j>=yStart) && (j<yStart+Picture_Height) && ((width - [...]

Hi, I haven't tried your method yet, but I have restored the original implementation method:
```
void EPD_7IN5B_V2_Display(const UBYTE *blackimage, const UBYTE *ryimage)
{
    UDOUBLE Width, Height;
    Width =(EPD_7IN5B_V2_WIDTH % 8 == 0)?(EPD_7IN5B_V2_WIDTH / 8 ):(EPD_7IN5B_V2_WIDTH / 8 + 1);
    Height = EPD_7IN5B_V2_HEIGHT;

    //send black data
    EPD_7IN5B_V2_SendCommand(0x10);
    for (UDOUBLE j = 0; j < Height; j++)
    {
        for (UDOUBLE i = 0; i < Width ; i++)
        {
            EPD_7IN5B_V2_SendData(blackimage[i + j * Width]);
        }
        
    }
    //EPD_7IN5B_V2_SendCommand(0x92);

    //send red data
    EPD_7IN5B_V2_SendCommand(0x13);
    for (UDOUBLE j = 0; j < Height; j++)
    {
        for (UDOUBLE i = 0; i < Width; i++)
        {
            EPD_7IN5B_V2_SendData(~ryimage[i + j * Width]);
        }
        
    }
    EPD_7IN5B_V2_TurnOnDisplay();
}
```
And added settings in the initialization function it works:
```

    //PANNEL SETTING: KW=3f, KWR=2F, BWROTP=0f, BWOTP=1f
    EPD_7IN5B_V2_Send(0x00, STR_CCH("\x0F"));

    //resolution setting: source=800, gate=480
    EPD_7IN5B_V2_Send(0x61, STR_CCH("\x03\x20\x01\xE0"));

    //VCOM AND DATA INTERVAL SETTING
    EPD_7IN5B_V2_Send(0x15, STR_CCH("\x00"));

    //VCOM AND DATA INTERVAL SETTING
    EPD_7IN5B_V2_Send(0x50, STR_CCH("\x11\x07"));

    //TCON SETTING
    EPD_7IN5B_V2_Send(0x60, STR_CCH("\x22"));

    //Gate/Source Start Setting: 800*480
    EPD_7IN5B_V2_Send(0x65, STR_CCH("\x00\x00\x00\x00"));
```
I think it's mainly the `0x61` and `0x65` command that works. Do you think I understand correctly?