waveshareteam/e-Paper #234: Paint_DrawRectangle Problem

MingYuanJeng, 2022-04-13:
參考此份原始碼測試 e-Paper/STM32/STM32-F103ZET6/User/GUI/GUI_Paint.c
問題測試代碼
Paint_NewImage(BaseImage, 800, 480, 0, WHITE);
Paint_Clear(RED);//先把整張圖畫成紅色
Paint_DrawRectangle(0, 0, 800, 480, BLACK, DOT_PIXEL_1X1, DRAW_FILL_FULL)//再把整張圖畫成黑色
參考圖片最底部的地方並沒有被畫成黑色
請問該如何解決呢?
謝謝
[image]
;

MingYuanJeng, 2022-04-15:
Refer to this source code e-Paper/STM32/STM32-F103ZET6/User/GUI/GUI_Paint.c
problem test code：
Paint_NewImage(BaseImage, 800, 480, 0, WHITE);
Paint_Clear(RED);//First make the whole picture red
Paint_DrawRectangle(0, 0, 800, 480, BLACK, DOT_PIXEL_1X1, DRAW_FILL_FULL);//Second step make the whole picture black
Why is the bottom part of the picture not painted black?
How to solve it please?
thanks

MingYuanJeng, 2022-04-18:
Does anyone know about this issue?

SSYYL, 2022-05-17:
I tested and the GUI worked.

Take a look at the complete code (from creating a BaseImage to refreshing the screen)

Note that STM32103ZET6 only has 64K RAM and may run out of memory to cause instability.

MingYuanJeng, 2022-05-18:
Hi SSYYL：
I am not testing on a microprocessor platform, I am testing GUI functions on a computer, so there will be no shortage of memory issues.