waveshareteam/e-Paper #239: Add examples for 7in5 V3 boards

CyanBlob, 2022-05-16:
I've got a couple of 7in5 V3 displays that I haven't been able to get working. I've tried modifying the V2 examples to match the specification from here https://www.waveshare.com/w/upload/8/8c/7.5inch-e-paper-b-v3-specification.pdf with no luck.

Can we get a 7in5 V3 example? These V3 displays are hard to use otherwise since there don't seem to be libraries that support them yet.

kiasarecool, 2022-05-17:
I can not get the screen to work either, only with the epd_7in5_test.py can I get anything... but it is just a black box.

SSYYL, 2022-05-17:
If you buy black and white screen, use this code: [epd_7in5_V2](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/examples/epd_7in5_V2_test.py)
If you buy black, white and red screen, use this code: [epd_7in5b_V2](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/examples/epd_7in5b_V2_test.py)

7.5inch e-Paepr (B) V3 uses the same program as V2.
Forgive me for forgetting to add the V3 version information to the Wiki.

CyanBlob, 2022-05-18:
I can confirm that the V2 demo works for me V3 board. Thanks for the tip! I do think it would be nice to either add a V3 demo (even if it's just a copy of the V2 code), or to add documentation explaining that the V3 should be backwards compatible with V2.

Regardless, I will go ahead and close this issue.

Thanks!

closed 2022-05-18 by CyanBlob