waveshareteam/e-Paper #240: sleep() should not call edpconfig.module_exit()

lordpengwin, 2022-05-28:
I'm using epd7in5_V2.py in a long running program that updates the display every 10 minutes. According to the documentation you should call sleep() when not updating the display and init() to wake it back up. This does not work because sleep() calls edpconfig.module_exit() which leaves the program in a bad state.

SSYYL, 2022-06-21:
Hi @lordpengwin 

I don't understand 'bad state', module_exit() simply runs cleanup() on a few necessary pins.

lordpengwin, 2022-06-21:
If you have an application that runs constantly in a loop and at the end of the loop it calls sleep() to put the display to sleep (as recommended by Waveshare support), the next time through the loop it fails with an error that says something about a pin not being set to output or something like that (i no longer have access to the display to test). Commenting out the call the module_exit() resolved the problem for me.