waveshareteam/e-Paper #249: Request: EPD_7in5_V2  Partial refresh

Sbatushe, 2022-08-04:
Hello, it's possible to implement partial refresh on a 7 inch display? Which steps are needed to implement this?

Sbatushe, 2022-09-15:
Hi, i implemented this some months ago, now i paused the project, but i will make a merge request. My partial refresh works, but i don't know if it's safe for the display.

SSYYL, 2022-09-15:
Please submit PR. I'd like to see it.

chellmann, 2022-10-07:
@Sbatushe it would be great if you could share your work with us :-)

Sbatushe, 2022-10-10:
I'm not a Github expert, how i can i send the code? Merge request? I need to power on My Pi and check the code.

chellmann, 2022-10-10:
First fork the Repo, place your altered files in there and then commit and push it to Github. Then you could provide your Fork here and may fill a Pull request.

You could also tell us how you achieved the partial refresh and post the code you changes as comment in this issue.

Anything would be really appreciated!

Sbatushe, 2022-10-13:
Ok ok, only problem is to clean up my project, i'll upload it when it's a bit cleaner, however to obtain the partial refresh i changed some parameter in the init phase (probably the clock speed) and then removed unnecessary steps in the lut, that's it

Sbatushe, 2022-11-28:
Sorry for the wait, this is my modified library. It's a mess and it uses 4inV2 code mixed with 7inV2

```
import logging
from . import epdconfig

# Display resolution
EPD_WIDTH       = 800
EPD_HEIGHT      = 480

class EPD:

    lut_vcom0 = [ 
    0x00, 0x17, 0x00, 0x00, 0x00, 0x02,        
    0x00, 0x17, 0x17, 0x00, 0x00, 0x02,        
    0x00, 0x0A, 0x01, 0x00, 0x00, 0x01,        
    0x00, 0x0E, 0x0E, 0x00, 0x00, 0x02,        
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,        
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,        
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    ]
    lut_ww = [
    0x40, 0x17, 0x00, 0x00, 0x00, 0x02,
    0x90, 0x17, 0x17, 0x00, 0x00, 0x02,
    0x40, 0x0A, 0x01, 0x00, 0x00, 0x01,
    0xA0, 0x0E, 0x0E, 0x00, 0x00, 0x02,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    ]
    lut_bw = [
    0x40, 0x17, 0x00, 0x00, 0x00, 0x02,
    0x90, 0x17, 0x17, 0x00, 0x00, 0x02,
    0x40, 0x0A, 0x01, 0x00, 0x00, 0x01,
    0xA0, 0x0E, 0x0E, 0x00, 0x00, 0x02,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    ]
    lut_wb = [
    0x80, 0x17, 0x00, 0x00, 0x00, 0x02,
    0x90, 0x17, 0x17, 0x00, 0x00, 0x02,
    0x80, 0x0A, 0x01, 0x00, 0x00, 0x01,
    0x50, 0x0E, 0x0E, 0x00, 0x00, 0x02,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    ]
    lut_bb = [
    0x80, 0x17, 0x00, 0x00, 0x00, 0x02,
    0x90, 0x17, 0x17, 0x00, 0x00, 0x02,
    0x80, 0x0A, 0x01, 0x00, 0x00, 0x01,
    0x50, 0x0E, 0x0E, 0x00, 0x00, 0x02,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    ]
    
    #-----------PARTIAL REFRESH HOMEMADE -----------------------------------------------------
    EPD_4IN2_Partial_lut_vcom1 =[
    0x00, 0x0E, 0x0E, 0x00, 0x00, 0x02,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
    ]

    EPD_4IN2_Partial_lut_ww1 =[
[... 204 lines cut ...]
        
    def Clear(self):
        self.send_command(0x10)
        for i in range(0, int(self.width * self.height / 8)):
            self.send_data(0x00)
            
        self.send_command(0x13)
        for i in range(0, int(self.width * self.height / 8)):
            self.send_data(0x00)
                
        self.send_command(0x12)
        epdconfig.delay_ms(100)
        self.ReadBusy()

    def sleep(self):
        self.send_command(0x02) # POWER_OFF
        self.ReadBusy()
        
        self.send_command(0x07) # DEEP_SLEEP
        self.send_data(0XA5)
        
        epdconfig.module_exit()
### END OF FILE ###

```

Cylindric, 2023-03-14:
@Sbatushe thanks for that code, it actually worked for me straight as it is. 

Do you know if it's still necessary or desirable to do any sort of full-screen refresh in the old way too? And if so, is there an easy way of calling the "old" `display` method? I'm finding that manually sending a full-black screen followed by a full-while screen is much slower than the old process.

Cylindric, 2023-03-16:
I take it back a bit. That modified lib doesn't always seem to work. I found that in many redraw scenarios it just results in a corrupted screen. I've not yet been able to work out the steps to reproduce though. Seems to be that I've drawn an image to the screen once just fine, but later changes result in only partial updates and then fading elements. I'll try and create a test-case not embedded in all my crap.

takosalad, 2023-10-03:
"For E-paper displays that support partial refresh, please note that you cannot refresh them with the partial refresh mode all the time. After refreshing partially several times, you need to fully refresh EPD once. Otherwise, the display effect will be abnormal, which cannot be repaired!"

Taken from https://www.waveshare.com/wiki/4.2inch_e-Paper_Module_(B)_Manual

Cylindric, 2023-10-03:
@takosalad that doesn't really relate though. I can't do any partial refreshes without subsequent draws corrupting. That wasn't the case with the old code.

va1m, 2024-01-14:
@SSYYL, @waveshare 
Does the screen support partial update physicaly at all?

mbartlett21, 2024-01-14:
The chip that is used to drive the display (GD7965) does have two bits of memory per pixel, so as long as you can get it into a mode where it reads from one bank of memory and sets the other to that (so it knows what the display is currently showing) that should work.