waveshareteam/e-Paper #257: epd1in54b (V1) dim red color and border around screen on RasPi 4B

ShadF0x, 2022-08-29:
Tried running RasPi epd1in54b demo for V1 display.

Red color is very dim, almost invisible. If I draw the same framebuffer as both red and black the picture gets better, but it's still nowhere near as vivid as I saw in other people's projects.

Also, the demo causes a thin black border to appear around the edge of the panel.

Documentation on the panel says that both the contrast and the border can be controlled via either VCOM or "Border Waveform Control register", but I can't find any info on *how* to actually do it.

cjs30, 2022-09-13:
Have you found how to do this yet? I will have a play this morning.

ShadF0x, 2022-09-13:
Nope.

After reading the [specs](https://www.waveshare.com/w/upload/9/9e/1.54inch-e-paper-b-v2-specification.pdf) some more, I found something that's mentions Border Waveform Control on page 25.

Seems like the doc references to this: https://github.com/waveshare/e-Paper/blob/dee10ea85d16df3e8879c31ad4f16d1925551307/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd1in54b_V2.py#L111

In theory, if you change 0x3C to something else (say, to 0x38 or 0x3F), it _should_ affect the red channel. 

That said, I didn't have time to test it yet, and I have no idea what most of the abbreviations they use mean (even after googling), so I might be dead wrong.

w3p706, 2022-11-08:
I was able to change the border of my display to black with

```
_writeCommand(0x3C); // BorderWavefrom
_writeData(0x00);
```

instead of the orginal which gave a white border
```
_writeCommand(0x3C); // BorderWavefrom
_writeData(0x05);
```

but i didn't found a description of the values for this command.