waveshareteam/e-Paper #259: Missing example EPD_7in5bc_V2

Ziesie1, 2022-09-10:
Hello,
which example in the folder "Arduino/" is the correct version for [this](https://www.waveshare.com/product/displays/e-paper/epaper-1/7.5inch-e-paper-hat-b.htm) e-paper?

As per the [documentation](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B)_Manual#Resources) it should be the "EPD_7in5bc_V2" example. The name is also listed in the [Arduino/README.md](https://github.com/waveshare/e-Paper/blob/master/Arduino/README.md) file but no corresponding folder could be found.

The example in [epd7in5b_V2](https://github.com/waveshare/e-Paper/tree/master/Arduino/epd7in5b_V2) didn't work for me. It hangs during initialization in WaitUntilIdle().
I have the 7.5 inch model V3 and the HAT Rev2.2.

SSYYL, 2022-09-15:
[epd7in5b_V2](https://github.com/waveshare/e-Paper/tree/master/Arduino/epd7in5b_V2)  is applicable to your screen.
Note that you need to select "4-line SPI".

Ziesie1, 2022-09-15:
Thank you for the response.
"4-line SPI" is selected and all connections were checked twice.

With a logic analyzer I can confirm the following behavior:
After the command POWER ON (04h) the busy pin of the display goes low but never comes high again. This is why the mcu hangs in the `WaitUntilIdle()` loop.
What does this mean? 

I am also wondering why in the [example epd7inb_V2](https://github.com/waveshare/e-Paper/blob/master/Arduino/epd7in5b_V2/epd7in5b_V2.cpp#L109) the command 71h is being sent during `WaitUntilIdle()`.
This command is not listed in the  [datasheet](https://www.waveshare.com/w/upload/8/8c/7.5inch-e-paper-b-v3-specification.pdf)?

matthias-bs, 2022-10-14:
Hello,
I am encountering the same problem as @Ziesie1. I am trying to use the same display with the [WaveShare e-Paper ESP32 Driver Board](https://www.waveshare.com/e-paper-esp32-driver-board.htm).

matthias-bs, 2022-10-14:
I found the following hint in the FAQ: [Question:Why is the BUSY pin always busy?](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B)_Manual#accordion13), but the answer does not really help me to understand the problem.

matthias-bs, 2022-10-14:
Another weird thing: normally, the ESP32's HSPI interface is wired as follows (see https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino):
```
SCLK - GPIO14
MOSI - GPIO13
```
but on the WaveShare e-Paper ESP32 Driver Board - according to the [Wiki](https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board#Documentation) pin list and the [schematic](https://www.waveshare.com/w/upload/8/80/E-Paper_ESP32_Driver_Board_Schematic.pdf) - we have:
```
SCLK - GPIO13
MOSI - GPIO14 (SDI)
```

matthias-bs, 2022-10-14:
And maybe because of this, SPI transfers are implemented by bit-banging in the ESP32 Driver Board examples (see DEV_Config.c).

Ziesie1, 2022-10-15:
My display is probably broken.
I also tried the raspberry-pi implementation [(link)](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B)_Manual#Raspberry_Pi), but here I had the same problem with the busy pin.

matthias-bs, 2022-10-15:
I tried the MicroPython version from https://github.com/tanahy/micropython-waveshare-epaper today. I tweaked the pin configuration a little bit to adapt to the [WaveShare e-Paper ESP32 Driver Board](https://www.waveshare.com/e-paper-esp32-driver-board.htm) and got the following results:
Ch 0 - brown: SCLK
Ch 1 - red: nCS
Ch 2 - orange: SDI
Ch 3 - green: nBusy
Ch 4 - yellow: DC (i.e. Data/nCommand)
Ch 5 - blue: nReset
"3SPI 0" is the decoded 3-wire SPI bus.

Reset pulse:
[image]

Then we have got the begin of the init sequence: 
1. Command POWER_SETTING (0x01)
2. Data 0x07
3. Data 0x07
[image]

...
4. Data 0x3F
5. Data 0x3F
6. Command POWER_ON (0x04)
After the POWER_ON command, nBusy turns low and remains low.
[image]

Repeat:
   7. Command GET_STATUS (0x71)
   8. Poll nBusy
[image]

Any ideas?

matthias-bs, 2022-10-15:
test.py:
```
"""
	Example for 7.5 inch black & white Waveshare E-ink screen, V2
	Run on ESP32
"""

import epaper7in5_V2
from machine import Pin, SPI, SoftSPI

sck = Pin(13)
miso = Pin(12)
mosi = Pin(14)
dc = Pin(27, Pin.OUT)
cs = Pin(15, Pin.OUT)
rst = Pin(26)
busy = Pin(25, Pin.IN)
spi = SoftSPI(baudrate=20000000, polarity=0, phase=0, sck=sck, miso=miso, mosi=mosi)

e = epaper7in5_V2.EPD(spi, cs, dc, rst, busy)
e.init()

w = 800
h = 480
x = 0
y = 0

# --------------------
[...]
```

matthias-bs, 2022-10-15:
I also tried to change the baudrate, switched the display type (A/B) and used an external 3.3V power supply - to no avail.

matthias-bs, 2022-10-16:
I just found the following PR: https://github.com/waveshare/e-Paper/pull/193
After changing EPD_7in5b_V2.cpp as follows, I got something on the display!
```
/******************************************************************************
function :	Wait until the busy_pin goes LOW
parameter:
******************************************************************************/
void EPD_7IN5B_V2_WaitUntilIdle(void)
{
    Debug("e-Paper busy\r\n");
    volatile unsigned char busy;
        int iter = 0;
	do	{
                //EPD_7IN5B_V2_SendCommand(0x71);
		busy = DEV_Digital_Read(EPD_BUSY_PIN);
		busy =!(busy & 0x01);
                DEV_Delay_ms(100);
                if (iter++ > 150)
                    break;
	} while(busy);
	DEV_Delay_ms(200);      
        Debug("e-Paper busy release\r\n");
}
```
So, instead of waiting for the Busy signal to become de-asserted, the wait loop left after a timeout. And the `SendCommand(0x71)` (`GET_STATUS`) does not seem to be necessary.

matthias-bs, 2022-10-18:
I ordered a second display (this time 7.5 inch T7, i.e. black/white) and a second [WaveShare e-Paper ESP32 Driver Board](https://www.waveshare.com/e-paper-esp32-driver-board.htm).

Now I found out that
- My first [WaveShare e-Paper ESP32 Driver Board](https://www.waveshare.com/e-paper-esp32-driver-board.htm) is broken
- The example code "epd7in5b_V2-demo" from E-Paper_ESP32_Driver_Board_Code.7z works with the 7.5 inch black/white/red display (type DEPG0750RW) as stated above

Ziesie1, 2022-10-22:
That's how it was for me, too @SSYYL.

The new display is now running fine with the example code.

closed 2022-10-22 by Ziesie1