waveshareteam/e-Paper #263: Unable to draw with epd2in13_V3

Chriss316, 2022-09-23:
Hi,

I have a 2.13" V3 screen and I am unable to get epdpaint to output anything to the screen. When I upload to my arduino the screen flashes between black and white a couple of times then goes to sleep. I have used the basic example code included in epd2in13_V3 to test the screen and that code works fine.

Below is the code I am using, not sure if I am missing something. All the examples I have looked at seem to be for V1 and V2 screens and use functions not available in epd2in13_V3. I have also tried using epd2in13_V2 and epd2in13d, neither of which work.

```
#include <SPI.h>
#include "epd2in13_V3.h"
#include "epdpaint.h"
#include "imagedata.h"

#define COLORED     0
#define UNCOLORED   1

unsigned char image[1024];
Paint paint(image, EPD_WIDTH, EPD_HEIGHT);
Epd epd;

void setup()
{
  epd.Init(FULL);
  epd.Clear();
  
  paint.Clear(UNCOLORED);

  paint.DrawStringAt(0, 0, "Hello World!", &Font16, COLORED);
//  paint.DrawFilledCircle(0, 0, 20, COLORED);
//  paint.DrawCharAt(0, 0, 'A', &Font16, COLORED);
//  paint.DrawHorizontalLine(0, 0 , 100, COLORED);
  
  epd.Display(paint.GetImage());

  epd.Sleep();
}
void loop() {
}
```

Chriss316, 2022-09-23:
I've modified the code slightly and now instead of a blank screen I am getting noise as the output.

```
unsigned char image[1024];
Paint paint(image, 0, 0); // <-- Changed the width/height to 0 and moved into the setup
Epd epd;

void setup()
{
  epd.Init(FULL);
  epd.Clear();
  
  paint.Clear(UNCOLORED);
  paint.SetWidth(EPD_WIDTH);     // <-- SetWidth/Height here instead
  paint.SetHeight(EPD_HEIGHT);
  
  paint.DrawStringAt(5, 5, "Hello World!", &Font16, COLORED);
  
  epd.Display(paint.GetImage());
  epd.Sleep();
}
```
[image]

SSYYL, 2022-09-29:
Hello, does the direct run routine display correctly? Looks like Paint didn't port well. I'll check later.

Chriss316, 2022-10-06:
Ok, I got some time to take a look at the code again. Starting with the basics I’ve run a condensed version of the example code, which produces the expected result of an image of flowers. (Image 1).
```

#include <SPI.h>
#include "epd2in13_V3.h"
#include "imagedata.h"

Epd epd;

void setup()
{
  Serial.begin(115200);
  epd.Init(FULL);
  epd.Display(IMAGE_DATA);
  epd.Sleep();
}

void loop(){}
```

I can also draw to the screen by creating a custom char array, in this case I just quickly threw together a pattern that would render alternating black and white stripes across the screen (Image 2). Its effectivly just doing the same as IMAGE_DATA above. 

```
#include <SPI.h>
#include "epd2in13_V3.h"
#include "imagedata.h"

Epd epd;
const unsigned char IMAGE_1[] PROGMEM = {
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
  0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,0X00,0XFF,
[... 269 lines cut ...]
  epd.SetFrameMemory(paint.GetImage(), 16, 60, paint.GetWidth(), paint.GetHeight());

  paint.Clear(UNCOLORED);
  paint.DrawCircle(32, 32, 30, COLORED);
  epd.SetFrameMemory(paint.GetImage(), 72, 60, paint.GetWidth(), paint.GetHeight());

  paint.Clear(UNCOLORED);
  paint.DrawFilledRectangle(0, 0, 40, 50, COLORED);
  epd.SetFrameMemory(paint.GetImage(), 16, 130, paint.GetWidth(), paint.GetHeight());

  paint.Clear(UNCOLORED);
  paint.DrawFilledCircle(32, 32, 30, COLORED);
  epd.SetFrameMemory(paint.GetImage(), 72, 130, paint.GetWidth(), paint.GetHeight());
  epd.DisplayFrame();
}

void loop() {
}
```

I took a quick look over epdpaint.cpp in both epd2in13 and epd2in13_V3 and they look to be identical, so I am wondering if something has gone wrong with epd.Display(). Perhaps pgm_read_byte is not being correctly iterated over.

[image]
[image]
[image]