waveshareteam/e-Paper #264: No info for 7.3" 4-colour display

berlingozzo, 2022-09-28:
hi
The examples for the 4-colour 7.3" display are very limited, just clearscreen and image load. The Paint class is not ported. The header file doesn't report all driver commands. The wiki doesn't have proper datasheet for the driver of this panel.
Same with all the 4-colour panels.

Can we have more material on these displays please?
thanks

SSYYL, 2022-09-29:
Hello, now as well as normal upload data manual. [Datasheet](https://www.waveshare.com/w/upload/9/99/7.3inch_e-Paper_%28G%29_Specification.pdf)
Due to lack of memory, Paint class has not been ported. You can refer to the Raspberry Pi program.

niels747, 2022-10-02:
How can I use the paint class on Arduino compatible microcontroller with enough memory, like arduino due or esp32?

niels747, 2022-10-06:
Because nothing was available, I modified the Paint class to support the 4 color display myself. Works like a charm!
Contact me if you want the code.
No i'm just searching for a fast way to create images in this 4 color 2 bit per pixel format...

csloz, 2022-11-15:
@niels747 Feel like sharing the code?

niels747, 2022-11-15:
@csloz This is what I am working on right now, it gets weather data from an API and displays it. I am using an ESP32, note that an atmega wont have enough ram, the framebuffer alone is 96kB. You can take from it what you need. https://drive.google.com/file/d/1lxj2B16aUH3BbRdJUYE4u6E77cw2h91z/view?usp=share_link
[image]

berlingozzo, 2022-11-16:
Thanks for sharing.
Given we are on github, can you please share it as code in github, rather
than a zip in google drive?
Many thanks and regards

sammcj, 2023-12-05:
@niels747 did you end up taking your work with the 7.3" any further?