waveshareteam/e-Paper #270: 7in5b V3 won't draw to screen

ashleahhill, 2022-10-24:
I have the Rev 2.2 board, which shouldn't have the 3.5 V / 5V issue and 7.5inch e-Paper HAT (B) with a "V3" sticker on the back of the display. I've tried using the `epd_7in5b_HD_test.py` and c equivalent as well as `epd_7in5b_V2_test.py` and c equivalent. 

The "HD" example runs, but nothing is drawn to the screen. All other examples are stuck on the `init` call. 

Is there a missing driver for version V3?

godzillafan868, 2022-12-01:
I too am having the same issue.  The most I can get to display on the screen is either having it go entirely red or entirely white.  I've used both the Python and C tests supplied by Waveshare, and multiple different tests:
```
epd_7in5bc_test.py
epd_7in5b_HD_test.py
epd_7in5b_V2_test.py
epd_7in5_HD_test.py
epd_7in5_test.py
epd_7in5_V2_test.py
```

JohannMrndn, 2022-12-01:
Exactly same here, I did many tests but none of them draw anything on the screen...

NHanser, 2023-01-21:
Same for me. I have a V3 sticker on the back and tried epd_7in5b_V2_test.py. The output seems correct, but nothing drawn on screen...
`python3 epd_7in5b_V2_test.py 
INFO:root:epd7in5b_V2 Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
INFO:root:1.Drawing on the Horizontal image...
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
INFO:root:2.Drawing on the Vertical image...
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
INFO:root:3.read bmp file
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
INFO:root:4.read bmp file on window
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
INFO:root:Clear...
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
INFO:root:Goto Sleep...
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5b_V2:e-Paper busy release
DEBUG:waveshare_epd.epdconfig:spi end
DEBUG:waveshare_epd.epdconfig:close 5V, Module enters 0 power consumption ...`