waveshareteam/e-Paper #276: E-paper 2in13_V3 not working properly when used via jumper wires

pakal, 2022-11-15:
I have a 2in13 V3, which works properly when used as a HAT.

But the hat causes static noise on all pins (which is abnormal :?), and causes MEMS recording to be full of noise.

So I need to connect the screen via jumper wires.

When I do that, the screen half-breaks : when running examples like epd_2in13_V3_test.py, only the transient flower image appear during refresh, but the screen stays blank at the end of the operation (when the flowers are supposed to remain displayed). And fonts/rectangles/triangles don't appear.

I have the same problem with my own code : when I display an image, it appears a bit during the refresh, then disappears and the display remains blank. As if the screen didn't have the "strength" to make the final display.

I tried lowering the SPI max_speed_hz, it didn't help.

How can I have the screen to work with jumper wire connection ?

bvevang, 2023-08-13:
Did you find a solution? I have the exact same problem.

pakal, 2023-08-16:
Alas not for jumper wires used with a raspberry pi... that's really a pity, and the lack of support is disconcerting (other epaper providers are mostly dead). 

But I did manage to use the Epaper as a HAT, but cutting unused pins which were causing noise (some Vdd and ground IIRC).
Thus, the MEMS mic plugged on these separate pins didn't have static noise.

SD-ChrisD, 2024-03-08:
I have this issue as well. Using the wires instead of the direct HAT connection ruins everything and I have no idea why. Not a single solution all over the internet, really frustrating....