waveshareteam/e-Paper #290: Debug: e-Paper busy Raspberry PI Zero

SaschaHeyer, 2023-04-04:
Hi
I am using the EPD_7IN5_V2 with the Raspberry PI Zero and it only shows static noise. 
[image]

The following output is shown when the C examples are used
```
/***********************************/ 
Current environment: Raspbian
bcm2835 init success !!! 
/***********************************/ 
e-Paper Init and Clear...
Debug: e-Paper busy
Debug: e-Paper busy release
Debug: e-Paper busy
Debug: e-Paper busy release
1 S
Paint_NewImage
show window BMP-----------------
pixel = 800 * 480
Debug: e-Paper busy
```

when using the Python examples it shows 
```
INFO:root:epd7in5_V2 Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy release
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy release
INFO:root:1.Drawing on the Horizontal image...
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy release
INFO:root:2.Drawing on the Vertical image...
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy release
INFO:root:3.read bmp file
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy release
INFO:root:4.read bmp file on window
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy release
INFO:root:Clear...
DEBUG:waveshare_epd.epd7in5_V2:e-Paper busy
```

even though the output suggested it is indeed rendering something it still only shows the static noise.

Any suggestions?

SaschaHeyer, 2023-04-04:
HAT is set to 

Display Config: B - 0.47R
Interface Config: 1 - 3line SPI

benny3000, 2023-04-07:
This may be the a known issue with the waveshare displays. Idk since when, but the displays are stuck at "Debug: e-Paper busy"...  This might be interesting for you: https://github.com/waveshare/e-Paper/issues/30

SSYYL, 2023-04-10:
> 

Please use 4line SPI, all programs are based on it.

closed 2023-04-11 by SaschaHeyer