waveshareteam/e-Paper #297: Partial update over white background? It's possible?

danielcharrua, 2023-06-23:
Hello, I'm using epd2in9d.py but can't make partial refresh over white background. Have tried many different approaches but without success... Maybe it can't work on this display?

I'm using Raspi 3b+ and [296×128, 2.9inch flexible E-Ink display HAT for Raspberry Pi](https://www.waveshare.com/product/displays/e-paper/epaper-2/2.9inch-e-paper-hat-d.htm)

https://github.com/waveshareteam/e-Paper/blob/fc4f54334d087dae440e3e9ac2bf8d809dfcfb53/RaspberryPi_JetsonNano/python/examples/epd_2in9d_test.py#L77C5-L93C18

martinius96, 2023-07-02:
were you successful?

danielcharrua, 2023-07-03:
Hey @martinius96 

Y was successful using the next code, hope it helps others 💪

```
# # partial update
[logging.info](http://logging.info/)("5.show time...")
epd.init()    
epd.Clear()

time_image = Image.new('1', (epd.width, epd.height), 0)
time_draw = ImageDraw.Draw(time_image)
num = 0
while (True):
    time_draw.rectangle((10, 10, 120, 50), fill = 0)
    time_draw.text((10, 10), time.strftime('%H:%M:%S'), font = font24, fill = 255)
    newimage = time_image.crop([10, 10, 120, 50])
    time_image.paste(newimage, (10,10))  
    epd.DisplayPartial(epd.getbuffer(time_image))
    num = num + 1
    if(num == 10):
        break
```

closed 2023-07-03 by danielcharrua