waveshareteam/e-Paper #306: Wrong Elf Class error when trying example python on raspberry pi 4

rvijgen, 2023-10-11:
When I run the python example on a fresh up to date raspbian install on Raspberry Pi4 I get the following error

Traceback (most recent call last):
  File "/home/pi/shared/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in7b_test.py", line 11, in <module>
    from waveshare_epd import epd2in7b
  File "/home/pi/shared/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in7b.py", line 32, in <module>
    from . import epdconfig
  File "/home/pi/shared/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 238, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/home/pi/shared/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 116, in __init__
    self.SPI = ctypes.cdll.LoadLibrary(so_filename)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 454, in LoadLibrary
    return self._dlltype(name)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: /home/pi/shared/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/sysfs_software_spi.so: wrong ELF class: ELFCLASS64

LoieBensbane, 2023-10-13:
I'm experiencing the same issue with a fresh install of Bookworm on a raspberry pi zero W.

Linux raspberrypi 6.1.0-rpi4-rpi-v6 #1 Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv6l

Traceback (most recent call last):
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in13g_test.py", line 11, in <module>
    from waveshare_epd import epd2in13g
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py", line 32, in <module>
    from . import epdconfig
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 238, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 116, in __init__
    self.SPI = ctypes.cdll.LoadLibrary(so_filename)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 454, in LoadLibrary
    return self._dlltype(name)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: /home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/sysfs_software_spi.so: wrong ELF class: ELFCLASS64

jcolladosp, 2023-10-13:
Similar error on raspberry pi 4, last version of Raspberry pi os:
`
  File "/home/marco/democode/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_7in5b_V2_test.py", line 11, in <module>
    from waveshare_epd import epd7in5b_V2
  File "/usr/local/lib/python3.11/dist-packages/waveshare_epd-0.0.0-py3.11.egg/waveshare_epd/epd7in5b_V2.py", line 32, in <module>
    from . import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/waveshare_epd-0.0.0-py3.11.egg/waveshare_epd/epdconfig.py", line 238, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/waveshare_epd-0.0.0-py3.11.egg/waveshare_epd/epdconfig.py", line 119, in __init__
    raise RuntimeError('Cannot find sysfs_software_spi.so')
RuntimeError: Cannot find sysfs_software_spi.so`

chrisvanberg, 2023-10-14:
Similar issue on Raspberry Pi B+ running Raspbian 32bits

It seems that the `sysfs_software_spi.so ` is not compiled for 32 bit architecture.

jcolladosp, 2023-10-14:
> Similar issue on Raspberry Pi B+ running Raspbian 32bits
> 
> It seems that the `sysfs_software_spi.so ` is not compiled for 32 bit architecture.

But if you see `epdconfig.py` file, `sysfs_software_spi.so ` is only necesary for JetsonNano, no Raspberry Pi, so I dont know why but its geting the wrong system:
https://github.com/waveshareteam/e-Paper/blob/4822c075f5df714f88b02e10c336b4eeff7e603e/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py#L114

LoieBensbane, 2023-10-14:
> But if you see `epdconfig.py` file, `sysfs_software_spi.so ` is only necesary for JetsonNano, no Raspberry Pi, so I dont know why but its geting the wrong system: [...]

Lines 233-241 seem to be the answer to why it's getting the wrong system: JetsonNano is the else condition. It's not finding what it expects to at the path used to check that the code is being run on a raspberry pi.

jcolladosp, 2023-10-14:
> Lines 233-241 seem to be the answer to why it's getting the wrong system: JetsonNano is the else condition. It's not finding what it expects to at the path used to check that the code is being run [...]

I can confirm that on Raspberry Pi 4 with Raspberry Pi OS 64 bits, that file doesn't exist: `/sys/bus/platform/drivers/gpiomem-bcm2835`

LoieBensbane, 2023-10-14:
> I can confirm that on Raspberry Pi 4 with Raspberry Pi OS 64 bits, that file doesn't exist: `/sys/bus/platform/drivers/gpiomem-bcm2835`

I changed my epdconfig.py by cutting the path back to '/sys/bus/platform/drivers' so that it would at least proceed using the Rasberry Pi code.

A new error appears! (kindof... I'll explain after the Traceback)

INFO:root:epd2in13g Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd2in13g:e-Paper busy H
DEBUG:waveshare_epd.epd2in13g:e-Paper busy release
Traceback (most recent call last):
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in13g_test.py", line 23, in <module>
    epd.init()
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py", line 144, in init
    self.SetWindow()
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py", line 94, in SetWindow
    self.send_data(self.Source_BITS/256)
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py", line 81, in send_data
    epdconfig.spi_writebyte([data])
  File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 63, in spi_writebyte
    self.SPI.writebytes(data)
TypeError: Non-Int/Long value in arguments: b65691e0.

This is the same error I was seeing when running the same code on a freshly installed and prepared image of Bullseye (with an unmodified epdconfig.py). Bookworm removed the driver, but there's still another underlying problem.

It's worth noting that the C demo code works flawlessly on both. Unfortunately, I don't speak C.

LoieBensbane, 2023-10-15:
[Found the missing piece of the puzzle. ](https://www.reddit.com/r/raspberry_pi/comments/16nq8bm/waveshare_213_e_paper_flickering_on_busy/k3to0t4/) The python demo now runs for me without issue.

To summarize the changes made: 

At line 233 of `/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py
` I changed `/sys/bus/platform/drivers/gpiomem-bcm2835` to `/sys/bus/platform/drivers`

At line 94 of `/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py` I changed `/self.send_data(self.Source_BITS/256)` to `self.send_data(self.Source_BITS//256)`

At line 96 of `/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py` I changed `self.send_data(self.Gate_BITS/256)` to `self.send_data(self.Gate_BITS//256)`

jcolladosp, 2023-10-15:
> > > > > Similar issue on Raspberry Pi B+ running Raspbian 32bits
> > > > 
> > > > 
> > > > > 
> > > > 
> > > > 
> > > > > It seems that the `sysfs_software_spi.so ` is not compiled for 32 bit architecture.
> > > > 
> > > > 
> > > > But if you see `epdconfig.py` file, `sysfs_software_spi.so ` is only necesary for JetsonNano, no Raspberry Pi, so I dont know why but its geting the wrong system:
> > > > https://github.com/waveshareteam/e-Paper/blob/4822c075f5df714f88b02e10c336b4eeff7e603e/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py#L114
> > > 
> > > 
> > > Lines 233-241 seem to be the answer to why it's getting the wrong system: JetsonNano is the else condition. It's not finding what it expects to at the path used to check that the code is being run on a raspberry pi.
> > 
> > 
> > I can confirm that on Raspberry Pi 4 with Raspberry Pi OS 64 bits, that file doesn't exist: `/sys/bus/platform/drivers/gpiomem-bcm2835`
> 
> I changed my epdconfig.py by cutting the path back to '/sys/bus/platform/drivers' so that it would at least proceed using the Rasberry Pi code.
> 
> A new error appears! (kindof... I'll explain after the Traceback)
> 
> INFO:root:epd2in13g Demo INFO:root:init and Clear DEBUG:waveshare_epd.epd2in13g:e-Paper busy H DEBUG:waveshare_epd.epd2in13g:e-Paper busy release Traceback (most recent call last): File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in13g_test.py", line 23, in epd.init() File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py", line 144, in init self.SetWindow() File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py", line 94, in SetWindow self.send_data(self.Source_BITS/256) File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13g.py", line 81, in send_data epdconfig.spi_writebyte([data]) File "/home/username/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 63, in spi_writebyte self.SPI.writebytes(data) TypeError: Non-Int/Long value in arguments: b65691e0.
> 
> This is the same error I was seeing when running the same code on a freshly installed and prepared image of Bullseye (with an unmodified epdconfig.py). Bookworm removed the driver, but there's still another underlying problem.
> 
> It's worth noting that the C demo code works flawlessly on both. Unfortunately, I don't speak C.

What python version are you using?
If you are using Raspberry Pi you only need to change the last ifs to this:
`implementation = RaspberryPi()`

LoieBensbane, 2023-10-15:
> What python version are you using? 
I'm using Python 3.11.2 

> If you are using Raspberry Pi you only need to change the last ifs to this: `implementation = RaspberryPi()`
That is much more elegant. Thank you.

mendhak, 2023-10-15:
Unfortunately this seems to break [my project](https://github.com/mendhak/waveshare-epaper-display) as I have been referencing this repo as a module, to allow users to use different Waveshare displays. To be specific, using it with the [October 10th Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/).  

It would be appreciated if a waveshare team member could give some attention to this issue and implement a more reliable fix on the epdconfig instead, so users don't have to edit a file to get the waveshare code working with latest Raspberry Pi OS. Sorry not sure who that would be @SSYYL @kmanc @waveshare ?

mendhak, 2023-10-15:
For reference, here is how Adafruit detect Raspberry Pi: https://github.com/adafruit/Adafruit_Python_GPIO/blob/a12fee39839665966bd124fd22588b2c87ced9d2/Adafruit_GPIO/Platform.py#L84-L113

This thread has some useful suggestions too, looking for the `Model:` line in `cat /proc/cpuinfo`: https://raspberrypi.stackexchange.com/a/139704/540

Or looking for the existence of the `/etc/rpi-issue` file: https://raspberrypi.stackexchange.com/a/140325/540

Marv2190, 2023-11-14:
> Unfortunately this seems to break [my project](https://github.com/mendhak/waveshare-epaper-display) as I have been referencing this repo as a module, to allow users to use different Waveshare displays. To be specific, using it with the [October 10th Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/). It's also not limited to Raspberry Pi 4, it does happen on Raspberry Pi Zero, Zero two, etc.
> 
> It would be appreciated if a waveshare team member could give some attention to this issue and implement a more reliable fix on the epdconfig instead, so users don't have to edit a file to get the waveshare code working with latest Raspberry Pi OS. Sorry not sure who that would be @SSYYL @kmanc @waveshareteam ?

Write them via Email, they dont look here very often - but via Email you get near instant answers.

rjpcasalino, 2023-12-05:
>But if you see epdconfig.py file, sysfs_software_spi.so is only necesary for JetsonNano, no Raspberry Pi, so I dont know why but its geting the wrong system:

@jcolladosp  yeah, I don't get that either but was throwing me off for a bit as I thought SPI wasn't active (and it wasn't since I was using NixOS) but switching to Rpi OS (bookworm) also threw this error with SPI enabled and rebooted...so I just commented out the bits that were trying to load up the JetsonNano stuff and :tada: it works like a dream

gregyone, 2024-02-17:
> [Found the missing piece of the puzzle. ](https://www.reddit.com/r/raspberry_pi/comments/16nq8bm/waveshare_213_e_paper_flickering_on_busy/k3to0t4/) The python demo now runs for me without issue.
>
> To [...]

-----
Thanks man, it saves my day !
Greg