waveshareteam/e-Paper #308: ModuleNotFoundError: No module named 'Jetson'

Marv2190, 2023-11-05:
File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in13d_test.py", line 11, in <module>
    from waveshare_epd import epd2in13d
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13d.py", line 32, in <module>
    from . import epdconfig
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 238, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 121, in __init__
    import Jetson.GPIO
ModuleNotFoundError: No module named 'Jetson'

Iam on Bookworm and followed the tutorial for linux:

sudo apt-get update
sudo apt-get install python3-pip
sudo apt-get install python3-pil
sudo apt-get install python3-numpy
sudo pip3 install RPi.GPIO
sudo pip3 install spidev

But iam getting ModuleNotFoundError: No module named 'Jetson'.

Can someone help? The C Tutorial and Demo working fine. - i need python...

TWelten, 2023-11-05:
hey
i'm facing the exact same issue.. I also have that 2in13d product like yours
please let us know if you managed to solve this issue :)
cheers  .t.w.

TWelten, 2023-11-05:
well, i found a way to avoid having to use Jetson module
DISCLAIMER : the following is not an appropriate solution, it is not fixing fundamentally the issue

Going in the file : **~/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py**
you can force the program to use the RaspberryPi() class. At the bottom, in the **else** condition, instead of having **implementation = JetsonNano()** change it for **implementation = RaspberryPi()**
That way I managed to get the demo working fine on my raspberry pi 4 (python3)

Yet i'd like better having a nicer fix to offer..
cheers .t.w.

robweber, 2023-11-17:
This is likely the same issue as #306 

The underlying issue is how the `epdconfig` class is trying to detect which hardware you're using, which fails on Raspberry Pi OS 12. There is a fix #307 but it has yet to be merged.

txoof, 2024-01-07:
I think this might be the same issue as #313. It looks like several modules are still trying to import RPi.GPIO when it is no longer needed. 

Impacted modules:

- epd2in13d
- epd2in9d
- epd4in01f
- epd4in2_V2

I've fixed this in PR #324, but don't have these models to test with; they get flagged by my general-purpose (epdlib)[https://github.com/txoof/epdlib) as being incompatible. I've narrowed down the problem to the import and that appears to fix the problem.