waveshareteam/e-Paper #313: Could not determine Jetson model

mudalu, 2023-11-26:
硬件环境：树莓派4B
屏幕：7.5inch e-Paper HAT版本
系统环境：Linux raspberrypi 6.1.0-rpi6-rpi-v8
Python环境：Python 3.11.2

运行e-Paper的epd_7in5b_V2_test.py，报错：Could not determine Jetson model

 原本报错Can't find Jetson.GPIO model，pip安装Jetson.GPIO后又报错：Could not determine Jetson model

shhds, 2023-12-21:
使用最新的程序即可

txoof, 2024-01-07:
This appears to be a problem with several modules due to importing the `RPi.GPIO` module directly from the module. See this example in the [2in13d version](https://github.com/waveshareteam/e-Paper/blob/6db502f3dd3a8a5760ea127cf1d142195165f64c/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13d.py#L31-L34)

Commenting out the `import RPi.GPIO` line resolves this.

The following modules appear to be impacted by this: 

* epd2in13d
* epd2in9d
* epd4in01f
* epd4in2_V2

```
import logging
from . import epdconfig
from PIL import Image
# the following line is causing this.
import RPi.GPIO as GPIO

# Display resolution
EPD_WIDTH       = 104
EPD_HEIGHT      = 212
```

txoof, 2024-01-07:
PR #324 should resolve this issue for all of the impacted modules.

@shhds: Do you prefer PRs pushed against the master or Development branch?