waveshareteam/e-Paper #322: 2.7inch display V2 IndexError: list index out of range error on raspberry

kriskut08, 2023-12-31:
when calling the display_4Gray function the program crashes
it also happens with the example
```
Traceback (most recent call last):
  File "/home/raspberry/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in7_V2_test.py", line 139, in <module>
    epd.display_4Gray(epd.getbuffer_4Gray(Limage))
  File "/usr/local/lib/python3.11/dist-packages/waveshare_epd-0.0.0-py3.11.egg/waveshare_epd/epd2in7_V2.py", line 450, in display_4Gray
    temp1 = image[i*2+j]
            ~~~~~^^^^^^^
IndexError: list index out of range
```

shhds, 2024-01-05:
In e-Paper\RaspberryPi_JetsonNano\python\lib\waveshare_epd\epd2in7_V2.py, change line 447 and 48000 in line 479 to 5808. The problem is thought to have been accidentally corrected by an engineer when adding a new screen. We will fix it in the next app update

kriskut08, 2024-01-06:
Thank you :D

brunojar, 2024-03-15:
When execute the: python btctcker.py 
happens with the example

INFO:root:Getting Data
ERROR:root:name 'epd2in7_V2' is not defined
Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 593, in fullupdate
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 480, in display_image
    epd = epd2in7_V2.EPD()
NameError: name 'epd2in7_V2' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 691, in main
    lastcoinfetch = fullupdate(config, lastcoinfetch)
  File "/home/pi/btcticker/btcticker.py", line 603, in fullupdate
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 480, in display_image
    epd = epd2in7_V2.EPD()
NameError: name 'epd2in7_V2' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 713, in <module>
    main()
  File "/home/pi/btcticker/btcticker.py", line 702, in main
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 480, in display_image
    epd = epd2in7_V2.EPD()
NameError: name 'epd2in7_V2' is not defined