waveshareteam/e-Paper #334: [HELP] WaveshareV4 not drawing on PI 02W (Device Busy)

aalejandrofer, 2024-03-06:
**Hardware**
- Pi Zero 2 W
- Waveshare 2.13inch e-Paper HAT v4

**Describe the bug**
When launching Pwnagotchi the web interface is up and running but the waveshare screen does not draw any images. 
- I have tested the screen on a pi4b and the image draws, I have texted the waveshare team but no reply wanted to see if anyone here encountered the same problem. 
- [Not Tested] I have heard that it has something to do with the screen being connected using the headers instead of wired. I want to avoid wiring it.

**Config**
ui.display.enabled = true
ui.display.type = "waveshare_4"
ui.display.color = “black”

**More Info**
```
ls /dev/spi* 
> /dev/spidev0.0  /dev/spidev0.1  /dev/spidev1.0  /dev/spidev1.1  /dev/spidev1.2

sudo pwnagotchi --debug
....
> AttributeError: 'Display' object has no attribute 'is_waveshare27inch'
[2024-01-26 23:19:32,646] [INFO] initializing waveshare v2in13_V4 display
INFO:root:BT-TETHER: Successfully loaded ...
[2024-01-26 23:19:32,673] [INFO] BT-TETHER: Successfully loaded ...
DEBUG:pwnagotchi.ui.hw.libs.waveshare.v2in13_V4.epd2in13_V4:e-Paper busy
[2024-01-26 23:19:32,715] [DEBUG] e-Paper busy

pwd
> /home/pi/e-Paper/RaspberryPi_JetsonNano/python/examples

python3 epd_2in13_V3_test.py
> INFO:root:epd2in13_V3 Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd2in13_V3:e-Paper busy
```

I have tested with waveshare_3 but the same

shhds, 2024-04-23:
I use the example code for testing is all right, you use the latest example code to see
Do you have other SPI devices using SPI at the same time

UnhappyCodings, 2024-05-11:
Hi! Im experiencing the same problem, V4 and V2 wont work at all on my 2.9inch EPaper module

I also can only epd2in9d and epd2in9b_V3 get to work on it.
Im using an Teensy 4.0 together with the Arduino IDE 2.3.2

UnhappyCodings, 2024-05-11:
Also seems like almost every library which includes "paint" does not work on my device