waveshareteam/e-Paper #335: lgpio.error: 'GPIO busy'

jayofelony, 2024-03-13:
I have waveshare V2 e-paper and when I run the python test file I get an error:

```
pi@jayofelony:~/e-Paper/RaspberryPi_JetsonNano/python/examples $ python3 epd_2in13_V3_test.py 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 408, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({'GPIO17', 'J8:11', 17, 'BCM17', 'BOARD11', '17', 'WPI0'}), pull='', row=6, col=1, interfaces=frozenset({'', 'spi', 'gpio', 'dpi', 'uart'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_2in13_V3_test.py", line 11, in <module>
    from waveshare_epd import epd2in13_V3
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd2in13_V3.py", line 32, in <module>
    from . import epdconfig
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 276, in <module>
    implementation = RaspberryPi()
                     ^^^^^^^^^^^^^
  File "/home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 52, in __init__
    self.GPIO_RST_PIN    = gpiozero.LED(self.RST_PIN)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 410, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```

jayofelony, 2024-03-15:
Apparently this has to do with the latest raspios Bookworm release. GPIO pins 8 and 17 are already being used.

jayofelony, 2024-03-15:
in `/boot/firmware/config.txt` add the following `dtoverlay=spi0-0cs` and it should work.

closed 2024-03-16 by jayofelony

PolkovnikSvet, 2024-09-07:
Please tell me which line exactly to add?

PolkovnikSvet, 2024-09-07:
I can't get the 4WD Smart Car Kit for Raspberry Pi to run. I have a Raspberry Pi5. I post the error that he writes below.

#pi@raspberrypi:~/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5 $ sudo python test.py ADC
Hardware not supported
Hardware not supported
/usr/lib/python3/dist-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: 'can not open gpiochip'
  warnings.warn(
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 408, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=15, name='GPIO22', names=frozenset({'WPI3', 'BCM22', 22, '22', 'BOARD15', 'GPIO22', 'J8:15'}), pull='', row=8, col=1, interfaces=frozenset({'', 'sdio', 'dpi', 'jtag', 'i2c', 'gpio'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5/test.py", line 73, in <module>
    from Ultrasonic import *
  File "/home/pi/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/Code/Server-pi5/Ultrasonic.py", line 8, in <module>
    sensor = DistanceSensor(echo=echo_pin, trigger=trigger_pin ,max_distance=3)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 827, in __init__
    super().__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 257, in __init__
    super().__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 243, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 79, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 410, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 101, in __init__
    GPIO.setup(self._number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address
#

Daidalos117, 2025-03-21:
hi @jayofelony , do you remember how did you solve please? I Have the exact same reecurring issue.

jayofelony, 2025-03-21:
> in `/boot/firmware/config.txt` add the following `dtoverlay=spi0-0cs` and it should work.

This