waveshareteam/e-Paper #336: RuntimeError: Failed to add edge detection

Coastline-3102, 2024-03-27:
Hello,

I am using a RPI 4, with [this](https://www.amazon.com/dp/B07Z25LWTS) display. Following the instructions [here](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_Manual#Working_With_Raspberry_Pi) I have been trying to get the demo to work. The c demo works without issue, but whenever I run ` epd_7in5_V2_test.py` I get an error which reads:
```
/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  warnings.warn(
Traceback (most recent call last):
  File "/home/steel/e-Paper/RaspberryPi_JetsonNano/python/examples/epd_7in5_V2_test.py", line 11, in <module>
    from waveshare_epd import epd7in5_V2
  File "/home/steel/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd7in5_V2.py", line 32, in <module>
    from . import epdconfig
  File "/home/steel/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 276, in <module>
    implementation = RaspberryPi()
                     ^^^^^^^^^^^^^
  File "/home/steel/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 56, in __init__
    self.GPIO_BUSY_PIN   = gpiozero.Button(self.BUSY_PIN, pull_up = False)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/input_devices.py", line 412, in __init__
    super().__init__(
  File "/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/mixins.py", line 417, in __init__
    super().__init__(*args, **kwargs)
  File "/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/input_devices.py", line 168, in __init__
    self.pin.when_changed = self._pin_changed
    ^^^^^^^^^^^^^^^^^^^^^
  File "/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/pins/__init__.py", line 471, in <lambda>
    lambda self, value: self._set_when_changed(value),
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 639, in _set_when_changed
    self._enable_event_detect()
  File "/home/steel/.cache/pypoetry/virtualenvs/e-paper-Yk_kAceY-py3.11/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 220, in _enable_event_detect
    GPIO.add_event_detect(
RuntimeError: Failed to add edge detection
```
Any ideas how to fix this?

Coastline-3102, 2024-03-28:
I have done some more work with it, and it seems like if I install the dependances with poetry, I have to run the demo as sudo for it to work: `poetry run sudo python epd_7in5_V2_test.py`

Is this expected behavior, or is there something I should tweak to allow me to avoid running this as sudo? I have noticed that the guide does not show running the python demo as sudo, which seems to imply that is not intended behavior?

Hantoniorl, 2024-04-06:
Hello @Coastline-3102. I'm trying to do the same, with the same screen but using a Zero 2W. And I get the very same error. Did you get it to work?

Coastline-3102, 2024-04-06:
@Hantoniorl in my case, running the script with `sudo` was the fix. Since I used poetry, the full command was `poetry run sudo python epd_7in5_V2_test.py`

If you are using pip (as they do in the manual) I believe the command would be `sudo python3 epd_7in5_V2_test.py`

Hope that helps!

Hantoniorl, 2024-04-06:
@Coastline-3102 Thanks! I was using sudo, but I think I messed up installing many other tests and stuff. So I'm going to try again from scratch.

shhds, 2024-04-23:
I use the example code for testing is all right, you use the latest example code to see

rbreckner, 2024-12-02:
Hey!

I ran into the same error with both the pip waveshare-epaper package and the examples version on Raspberry Pi Zero.

After not being able to install lgpio in my virtual environment, I finally tried running the example code with system python and it worked. No sudo required.

### Venv Solution
I compared system and venv `pip list` and solved it by enabling system site packages.

```
//.venv/pyvenv.cfg:

include-system-site-packages = true
```

You could also add `--system-site-packages` when creating a new venv to have the same effect.

### Poetry Solution (maybe)
Just looked it up, there is a similar setting for poetry, which could be enabled by

`poetry config virtualenvs.options.system-site-packages = true`. 

I haven't tried that, but the case seems very similar.