waveshareteam/e-Paper #339: "e-Paper busy" from Python examples

iandouglas, 2024-05-15:
Raspberry Pi Zero 2W

I've copied @jayofelony's epdconfig.py file and still get "e-Paper Busy" on several different screens.

Since I don't always know which script to run based on the hardware, I'm looping over all of the scripts for that screen size like

`for i in *2in13*py; do python3 $i; done`

but they're all telling me that the e-Paper is busy.

I've tried one 2.13" screen, which is marked as a v4 but was used in a project (pwnagotchi) as a v3, but now seems to not refresh any more. I've also tried two separate 2.7" screens with the 4 buttons, as well as a large 7.5" display.

I've seen in the open issues and closed issues that the C examples might work better to clear the busy bit, but I can't compile any of the scripts because of a missing "lgpio.h" file ... I've also tried the other three settings in the Makefile for USE_LIBRPI but I get an error of missing header files for all 4 of the options.

Is there something wrong with my RPi Zero 2 W maybe?

Note to @jayofelony if you see this, I've tried your pwnagotchi build 2.8.9 as well as aluminum-ice 1.8.5, and get the same behavior on this Pi. I'm starting to suspect the Pi.

jayofelony, 2024-05-15:
Open an issue on my GitHub if you will, with logs and commands used.

iandouglas, 2024-05-16:
On the newer OS of the pwnagotchi 2.8.9 build, I'm able to install the `liblgpio-dev` package and compile the `epd` script:

* 7.5 inch screen: e-Paper busy
* 2.7 screen with 4 btns: e-Paper busy
* 2.13 screen: e-Paper busy

running like this:

```bash
for i in epd2in13V2 epd2in13V3 epd2in13V4 epd2in13bc epd2in13bV3 epd2in13bV4 epd2in13d epd2in13g; do 
  make clean; 
  make EPD=$i; 
  sudo killall pwnagotchi-launcher pwnagotchi bettercap-launcher bettercap pwngrid-launcher pwngrid; 
  echo; echo; echo $i; echo; echo; 
  ./epd; 
  echo; echo; echo; 
done
```

emilyboda, 2024-08-27:
I am also having this issue.