waveshareteam/e-Paper #369: 7.3 e-paper 6 color Informations on commands

Schoumi, 2024-11-20:
Hi,

i've looked on many many of the wiki pages to find informations but i can't find what i'm looking for.

I've looked into the repository to see how the display work and i've found that it's seems to exist many commands that are not documented in the wiki or the datasheet.

for exemple if i look into init function: https://github.com/waveshareteam/e-Paper/blob/master/Arduino_R4/src/e-Paper/EPD_7in3e.cpp#L116

The code send different command 0xAA, 0x01, 0x00 with some data but don't know what these are for and what values means. And I don't like to not know what i'm doing.

Did i miss a datasheet somewhere? Could you show me this datasheet or upload the datasheet that contains explanation to all these commands?

Thank you

psiegl, 2025-04-05:
I had the same issue, and tried to reverse-engineer a bit. You might find some hints here: https://codeberg.org/psiegl/E-Ink_Spectra_E6/src/branch/eink-spectra-e6/src/epdflash/src

Schoumi, 2025-04-08:
@psiegl Oh that's great. How did you find meaningful information about undocummented command?
I see, you change in your code some sequence values (BTST for exemples) what does these change do? 
Thanks for your work

@waveshare would be great if you can share the full documentation.

psiegl, 2025-04-08:
oh well, a lot of googling :)

Ater weeks I accidentally stumbled across https://github.com/CursedHardware/epd-driver-ic , which holds a significant amount of specs for EPD timing controllers. However, the question was: which one is the right one. Is it even part of?
I searched further and found https://shop.pimoroni.com/search?q=inky+impression . Pimoroni is selling old ACeP as well as Specta 6 displays, where the provide code for

- ACeP 7.3'' code: https://github.com/pimoroni/inky/blob/main/inky/inky_ac073tc1a.py
- ACeP 4.0''/5.65'' code: https://github.com/pimoroni/inky/blob/main/inky/inky_uc8159.py

Look at the last one, it states UC8159 and interestingly there is a spec given: https://github.com/CursedHardware/epd-driver-ic/blob/master/UC8159c.pdf . By taking the UltraChip UC8159 as a blueprint, I checked now also all other specs given in this repo. There I learned, that EPD TCONs that have a similar "command set":

- ... only employ 0x06 as BTST (Booster Soft Start), there is never a 0x08 or 0x05 (as these have the same pattern as 0x06)
- ... that 0x61 must be TRES i.e. the resolution setting, because the data [ SOURCE Part 0, Source Part 1, GATE Part 0, Gate Part 1 ] matches.
- ... that 0x12 must be DRF, i.e. the display refresh. An EPD always has a SRAM (according to all specs), which one fills with the SPI stream. While DTM moves the SRAM content into the EPD.
- ... I just tried out the 0x17 the AUTO command which can do PON->DRF->POF(->DSLP) and it worked nicely: https://github.com/waveshareteam/e-Paper/pull/387
- ... and I tried out the 0x30 i.e. the PLL control because it was very interesting to see, if it behaves similar as given in other specs. And indeed it does, see also: https://github.com/waveshareteam/e-Paper/pull/387
- ... and so on.

My current overview of commands: https://codeberg.org/psiegl/E-Ink_Spectra_E6/src/branch/eink-spectra-e6/src/epdflash/include/EPD_commands.h The unusal states, that these commands are typically not given in other specs. 

Well, but we want the Spectra 6 spec right? I checked on [GooDisplay](https://www.good-display.com/product/89/), which sells these EPDs as well:

- Spectra 6 4.0'': https://www.good-display.com/product/532.html <- check out the Download section (STM32&ESP32 code & spec)
- Spectra 6 7.3'': https://www.good-display.com/product/533.html <- check out the Download section (STM32&ESP32 code & spec)
- Spectra 6 13.3'': https://www.good-display.com/product/559.html <- check out the Download section (ESP-IDF code example & spec)
- there are also 25.3'' and 31.5'' but not much more content.

So, my gut feeling right now is, that the Spectra 6 either has a newer UltraChip TCON, where I don't have seen a 100% matching spec yet. However, I am believing, that it is even likely to be a Fitipower one, but again just guesses.

Over time, I detected where the initial EPD spec repo got their specs from. GooDisplay provides them for download: https://www.good-display.com/companyfile/5/ thus, I retrieved them and got a patch ready: https://github.com/CursedHardware/epd-driver-ic/pull/1

So yes, I am with you. I would love to see that either Waveshare or GooDisplay or E-Ink would provide just a proper spec. And I would like to see, that Waveshare fixes their PCBs, because I stumbled at least with the Spectra 6 4.0'' one into the same issue here: https://www.martinhubacek.cz/blog/waveshare-epaper-hat-issues/

bmuessig, 2025-04-24:
@psiegl Thank you for your analysis. The datasheet of the UC8179c seems promising for the 7.3" 7 color e-ink display. The known registers seem to match it quite closely.