waveshareteam/e-Paper #372: Problems coming after restart or deep sleep with ESP32 (7.3inch E Ink Spectra 6)

NickCole99, 2024-12-28:
Hello,

after running the default driver code on ESP32 for the 7.3inch E Ink Spectra 6 if you restart the device or exit deep sleep the screen will take 45 seconds to update instead of the usual 12. Please could this be looked into as I've not been able to find a solution.

Put this at the end of the default driver code and the screen will take 45+ seconds on every update.
    esp_sleep_enable_timer_wakeup(20 * 1000000); //20 Seconds 
    esp_deep_sleep_start();

Thank you

Theagainmen, 2025-01-12:
I have the same problem, updates always take ~45 seconds. I've only had once a 12s refresh. How come?

NickCole99, 2025-01-13:
hello, in the code you need to reinitlise the display if it takes too long

static void EPD_7IN3E_TurnOnDisplay(void)
{
    int startTime = millis();

    EPD_7IN3E_SendCommand(0x04); // POWER_ON
    EPD_7IN3E_ReadBusyH();

    EPD_7IN3E_SendCommand(BTST2);
    EPD_7IN3E_SendData(0x6F);
    EPD_7IN3E_SendData(0x1F);
    EPD_7IN3E_SendData(0x17);
    EPD_7IN3E_SendData(0x49);

    EPD_7IN3E_SendCommand(0x12); // DISPLAY_REFRESH
    EPD_7IN3E_SendData(0x00);
    EPD_7IN3E_ReadBusyH();

    delay(50);
    
    int elapsedTime = millis() - startTime;
    if (elapsedTime > 24000) {
        Serial.println("Operation took too long! Initilising again");
        EPD_7IN3E_Init();
        delay(200);
    }

    EPD_7IN3E_Sleep();

    Serial.print("Display time: ");
    Serial.println(elapsedTime);
}

NickCole99, 2025-01-22:
@Theagainmen Did you find a solution? I'm extremely stuck and have spent many hours trying to solve it

Theagainmen, 2025-01-22:
@NickCole99 I sadly have not.. I'm also still trying. Let's keep each other posted!