waveshareteam/e-Paper #385: Pi Zero Crash on Init 7.5V2 Display with Rev2.3 driver hat

JOMI195, 2025-03-09:
Hey guys,

I am using the `Raspberry Pi Zero V1.1` to power the `Waveshare 7.5 V2` Display over the `Display Driver Hat Rev 2.3`. Software wise I am using the Python `epd7in5_V2.py` library.

First the images getting displayed perfectly for a couple of hours. Then the pi crashes always in the `epd.init()` again and again. It's very frustrating. 

I tested also with several of the original power cables and some different SD cards. When no display code is used, my pi runs the program smoothly for days. No problem from this side in my opinion.

Does somebody also had problems with this kind of setup recently?

Cheers.

JOMI195, 2025-06-15:
I found out that it was a broken driver hat. I ordered two driver hats and both of them had these problems. I ordered them directly from the waveshare shop in china. After i ordered new ones from another source, the displays and the drivers where working as supposed with no chrashes.

masingbackstage, 2025-06-16:
Yeah, I ordered mine from Waveshare directly, but fortunately, I quickly realised it wasn't my fault...

opravdin, 2025-09-01:
Fixed that, those 2 things helped:
1. Ribbon cable was incorrectly connected from factory. Connectors were upside down on the HAT side. Placed connector with ribbon pins facing pcb.
2. Replaced 3v3 power to separate step down converter from 5v since there are reports about insufficient power for 7.5 inch display from Raspberry voltage converter.